nodemcu/nodemcu-firmware #70: Reboot loop when low on heap space/ram and Wifi mode is set to wifi.STATION

xlfe, 2015-01-01:
May be related to #23 

### How to reproduce

#### make sure you have low heap space

My init.lua looks like:
```lua
tmr.alarm(0, 1000, 0, function() require("bmp085") end )
```
Which takes a fair amount of heap space:
```lua
NodeMcu 0.9.4 build 20141226  powered by Lua 5.1.4
> print(node.heap())
8152
```

#### Set wifi mode to STATION

```lua
wifi.setmode(wifi.STATION)
```

#### Restart - a reboot loop occurs

### How to fix

Re-flash esp_init_data_default.bin

I have tried fixing this by changing the wifi mode back to SOFTAP but this doesn't help

I haven't had a chance to dig into the code yet but will update here if I figure out what is going on...

NeiroNx, 2015-01-02:
Yes. Just now tested wifi.STATION - some times it reeboot loop, but some times it normaly works(after 2...5 reboots) - may be it depend on WPA keys generated during wifi auth. 
Next you will try in open mode to verify guess.

vowstar, 2015-01-03:
We are do our best to reduce the heap use, and SDK 0.9.4 is unstable, we are waiting 0.9.5.

vowstar, 2015-01-11:
@xlfe Could you test the latest version? Does it fixed?

xlfe, 2015-01-11:
Hi @vowstar yes sorry latest version is much improved and fixes this issue.

closed 2015-01-11 by xlfe