nodemcu/nodemcu-firmware #71: send function - question(help)

Hoksmur, 2015-01-01:
In module net function send has 2 parameters. So, first is string - not a problem. Second is callbac function.
How can I use it? https://github.com/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en#parameters-54
Possible - I can't catch idea, because Lua is new for me.  

 And - how to read file binary? file.write() present, file.seek() - present, file.read() - ?

nodemcu, 2015-01-02:
the callback is same as calling socket:on("sent", function).
file.read(n) or file.read(end_char), has been added to branch dev094 already.
will merge to master and prebuild bin recently.

Hoksmur, 2015-01-03:
Thank you.
Hello!
I found on the Wiki information about size limit node.output() - ~1400 byte. I know way put long html code in string value, but how call 'send' few times? Is it possible?
Telnet exapmle didn't help me.
Can you explain me it, or give me short and(or) the easiest sample?
I saked the same here: http://bbs.nodemcu.com/t/can-somebody-help-me-with-node-dot-output/82
 And - close tiket.

closed 2015-01-12 by Hoksmur