nodemcu/nodemcu-firmware #72: How to Sink output on GPIO Pin in Lua

clixx-io, 2015-01-02:
I'm trying to get Gpio Pin#4 to sink current for a motor driver.

How do I do that?

So far I have:
 gpio.mode(4, gpio.OUTPUT)
 gpio.write(4,gpio.HIGHT)

but that sources current and I need active-low or to be able to sink current.

[image]

NeiroNx, 2015-01-02:
I am now measure resistance between GPIO(pin #4) and power lines. When gpio.LOW on output pin it about 0.03 kOhm to GND, and when it gpio.HIGH it about 0.08 kOhm to VCC. it can be used both active-LOW and active-HIGH lines. When using active-low you may also set PULLUP enable.

clixx-io, 2015-01-10:
I could not duplicate those results.

Normally an additional software option like 'ACTIVE_LOW' is needed in the software to sink current.

closed 2015-01-23 by nodemcu