nodemcu/nodemcu-firmware #73: multiple Send() bug?

pipi61, 2015-01-04:
Hi!
this example work on 14.12.19, not work on 14.12.30 firmware
if append the all html lines to single string, and send single command then work.
Thank You

wifi.setmode(wifi.STATION) 
wifi.sta.config("SS","PP") 
if srv ~= nil then srv:close() end
counter=1
srv=net.createServer(net.TCP) 
srv:listen(80,
  function(conn) 
    conn:on("receive",
     function(conn,pl)   
	 outmsg=nil
	counter=counter+1
	print (pl)
	outmsg = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n".."Connection: close\r\n"
	conn:send(outmsg) 
	outmsg ="\r\n<!DOCTYPE HTML>\r\n<html>\r\n"
	conn:send(outmsg) 
	outmsg ="<h1>This is the header of the page</h1><br />"
	conn:send(outmsg) 
	outmsg="Count = "..tostring(counter).."<br />"
	conn:send(outmsg) 
	outmsg ="</html>\r\n"
     conn:send(outmsg) 
    counter=counter+1
    print (counter)
    conn:on("sent",function(conn) print("package sent, Heap = "..node.heap()) conn:close() end)
    end)
  end)
print("start")

nodemcu, 2015-01-05:
is this the problem that close the conn too fast?
on("sent") will called if some of data send successfully.

pipi61, 2015-01-05:
Hi!
This example work on 2014.12.19, not work on 2014.12.22 and above firmware.
if insert tmr.delay(100) don't work also
conn:on("sent",function(conn) print("package sent, Heap = "..node.heap()) tmr.delay(100) conn:close()
sdk bug 9.2->9.4?
don't work this demo also:
http://www.esp8266.com/viewtopic.php?f=19&t=661
---
what max size large string?
Thank You

closed 2015-01-08 by nodemcu

pipi61, 2015-01-11:
Thank You!
Working, tested on 2015.01.10 firmware

commit b8e578f7 (2015-01-08): Merge pull request #77 from nodemcu/dev095

fix #73, net.socket:send() issue, fix file.read() when 0xFF reached