nodemcu/nodemcu-firmware #74: OMA Lightweight M2M

4refr0nt, 2015-01-05:
Enabling support <a href="http://fr.slideshare.net/zdshelby/oma-lightweightm2-mtutorial">LWM2M</a> communcation protocol in NodeMCU firmware by using <a href="https://github.com/sbernard31/lualwm2m ">lualwm2m</a> (lualwm2m is a Lua binding for client side <a href="https://github.com/eclipse/wakaama">liblwm2m/wakaama library</a>)
Server side: <a href="https://github.com/jvermillard/leshan">Leshan</a> is a OMA Lightweight M2M server and client implementation

I can implements leshan server-side part of software (partial, only ESP8266 specific values) into <a href="https://github.com/4refr0nt/ESPlorer">ESPlorer</a> for fast and easy developing IoT applications based on ESP8266+NodeMCU+Lua

[image]

nodemcu, 2015-01-05:
I've google-ed some lwm2m, but still don't know how to get to the module behind NAT from internet.
lwm2m based on coap which is UDP.
do you have some more info of lwm2m, and compare to mqtt. pros&cons

4refr0nt, 2015-01-05:
So you planing use mqtt as default and only one M2M protocol in NodeMCU?

nodemcu, 2015-01-05:
not decided yet. maybe use MACRO to control? 
put coap,mqtt and net module together is not a good idea I think. eats up every bits of ram.
or maybe NodeMCU-COAP, NodeMCU-MQTT?

4refr0nt, 2015-01-05:
We needed simple way for control other device connected to ESP8266 GPIO, send telemetry data from sensor without lot of lua code. We needed M2M protocol integrated into NodeMCU firmware NOW. It may be mqtt or any other. May be new NodeMCU protocol... Long-time-developing will be killing NodeMCU.
P.S. See https://github.com/tuanpmt/esp_mqtt

nodemcu, 2015-02-11:
for the COAP support.
please checkout https://github.com/nodemcu/nodemcu-firmware/tree/coap
you have to build the bin from code.

nodemcu, 2015-03-18:
now coap module is merged to master.

closed 2015-03-18 by nodemcu

mjkoster, 2015-06-19:
"I've google-ed some lwm2m, but still don't know how to get to the module behind NAT from internet.
lwm2m based on coap which is UDP."

You can keep a UPD return tunnel open to a LWM2M device by sending a LWM2M registration update from the device to the server frequently enough to prevent the NAT from timing out (usually 60 seconds).

So set up the LWM2M registration update to occur every 30 seconds and the LWM2M server should be able to reach the device behind the firewall.

Cheers,

Michael

TingChinLin, 2016-01-17:
I have already  bind "https://github.com/sbernard31/lualwm2m" in nodemcu.
Where should i update my nodemcu firmware for lwm2m?