nodemcu/nodemcu-firmware #83: Program On The Go

tuanpmt, 2015-01-09:
I have an idea: long press a button to go to setup mode, in this mode, we run wifi ap and webserver with lightweight lua editor. We don't need any other tools except browser to run new lua firmware on esp8266.
Is it possible?

nodemcu, 2015-01-09:
yes, it's possible.
use node.key to register long press callback.

maybe run wifi softap+station mode, to the same ap with the computer with browser.
use ace editor as the code editor, and cdn for js and css lib.  so nodemcu only host html file?

tuanpmt, 2015-01-09:
I want to make it as a part of native nodemcu code (C code). In setup mode just run httpd and don't run lua. Is lua machine run file main.lua? So may i just load and write main.lua for m idea?

tuanpmt, 2015-01-09:
This is lightweight editor: http://jakiestfu.github.io/Behave.js/
When compress, it can embeded to esp8266

Hoksmur, 2015-01-10:
Take a look here [uLuaNode](https://github.com/Hoksmur/uLuaNode) Do you want this functional?

closed 2015-01-13 by tuanpmt