nodemcu/nodemcu-firmware #85: attempt to concatenate (a nil value)

dannyvai, 2015-01-09:
this code inside a file generated this : 

ip = wifi.sta.getip()
print("ip is : " ..ip)

the output was : 
PANIC: unprotected error in call to Lua API (connect.lua:26: attempt to concatenate global 'ip' (a nil value))
PANIC: unprotected error in call to Lua API (attempt to call a string value)

and then it crushes the server and restarts

pipi61, 2015-01-09:
This work on old version lua, if wifi not connected, the wifi.sta.getip() return 0.0.0.0,
the new version return nil. The nil value not printable...

dannyvai, 2015-01-09:
The problame is that it crashes and resets the whole thing. 
If you just do print(nil) or
a=nil
print(a)
It doesn't crash

vowstar, 2015-01-11:
We changed this API due to previous issue, which though it is not good to return 0.0.0.0 when have no ip.
Before use ip, you must check if you really get an ip. You can't connect a string with a value may nil directly.
if(ip~=nil)
    -- do something
end

closed 2015-02-11 by nodemcu