nodemcu/nodemcu-firmware #86: it seems the call to genflashbinv6 in tools/gen_appbin.py can just be removed on linux?

hamishcunningham, 2015-01-09:
Hi folks. Great project -- thanks!

I've managed to rebuild and flash nodemcu succesfully on Ubuntu 14.04, using the esp-open-sdk toolchain.

I think there are a couple of glitches in the scripts for Linux though... 
After the initial "make" I do

    cd app; gen_misc.sh

This creates a bunch of .bin files which I can then flash using one of the esptools; this seems to work, but the gen_misc.sh call invokes gen_appbin.py, which then calls genflashbinv6 -- which isn't available except as a binary for Windows.

So we get part way there, but not all the .bin files are generated. So then I use the method from https://github.com/esp8266/esp8266-wiki/wiki/Building to finish off:

    cd .output/eagle/debug/image
    esptool-ck/esptool -eo eagle.app.v6.out -bo eagle.app.v6.flash.bin -bs .text -bs .data -bs .rodata -bc -ec
    xtensa-lx106-elf-objcopy --only-section .irom0.text -O binary eagle.app.v6.out eagle.app.v6.irom0text.bin
    cp eagle.app.v6.flash.bin ../../../../../bin/
    cp eagle.app.v6.irom0text.bin ../../../../../bin/

And this gives me the full set of files that I can then flash to the chip like this:

    esptool.py --port /dev/ttyAMA0 write_flash 0x00000 eagle.app.v6.flash.bin 0x10000 eagle.app.v6.irom0text.bin 0x7c000 esp_init_data_default.bin 0x7e000 blank.bin

This works :-)  (though it was quite hard to figure out!).

So perhaps gen_appbin.py should just omit the genflashbinv6 call on linux, and gen_misc.sh do the above instead? Or perhaps there's a better way to take the results of the initial build and make them flashable?

Thanks!

Hamish

iabdalkader, 2015-01-10:
Had the same problem, it seems there are two versions of esptool ? anyway, one of them supports the following:
```
esptool.py elf2image eagle.app.v6.out
```
It generates the two binaries needed, I call it from the top Makefile with an output prefix ```-Obin/``` like this:

```
flash:
	-$(ESPTOOL) elf2image $(TARGET_OUT) -obin/
	-$(ESPTOOL) --port $(ESPPORT) write_flash 0x00000 bin/0x00000.bin 0x10000 bin/0x10000.bin
```

iabdalkader, 2015-01-10:
Need to add that some scripts should be called with python2.x

hamishcunningham, 2015-01-10:
Looks good, thanks, I'll try it!
Thanks, Hamish

sej7278, 2015-01-10:
@iabdalkader - i came across this method too, it seems none of the gen* scripts/binaries are needed anymore, we can just use the newer version of esptool.py (which comes with esp-open-sdk and is also in the nodemcu/tools directory - not sure a local copy is a good idea, a submodule would make more sense):

```
make
cd app/.output/eagle/debug/image
esptool.py elf2image eagle.app.v6.out
esptool.py write_flash 0x00000 eagle.app.v6.out-0x00000.bin 0x10000 eagle.app.v6.out-0x10000.bin
```

iabdalkader, 2015-01-10:
I agree, and btw, I didn't use the local copy, I used the latest one:
https://github.com/themadinventor/esptool

sej7278, 2015-01-10:
i made a PR which is a start, probably won't work with their nodemcu-flasher app and needs a way to pass in the port for windows, but it seems better than the endless stream of shell scripts, closed binaries and batch files.

closed 2015-01-10 by nodemcu