nodemcu/nodemcu-firmware #95: Add xmodem transfer from eLua

kmpm, 2015-01-10:
I would love to have a simpler way of doing such a common thing as uploading files.

eLua have implemented a protocol called xmodem which adds a _recv_ command to the shell.
I don't know if it's possible to include this as an optional thing here or if we are missing something neccesary.

I found some related files in the elua git
https://github.com/elua/elua

* elua/elua
* elua/src/shell/shell_recv.c
* elua/inc/xmodem.h
* elua/inc/xmodem.c

Hoksmur, 2015-01-11:
It is real to program in lua. See example in [uart.on](https://github.com/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en#uarton) description.

kmpm, 2015-01-11:
Made my own similar implementation for use in https://github.com/kmpm/nodemcu-uploader
```
function recv_block(d)
  if string.byte(d, 1) == 1 then
    size = string.byte(d, 2)
    if size > 0 then
      file.write(string.sub(d, 3, 3+size))
      uart.write(0,'\006')
    else
      uart.write(0,'\006')
      file.close()
      uart.on('data')
      uart.setup(0,9600,8,0,1,1)
    end
  else
    uart.write(0, '\021' .. d)
    uart.setup(0,9600,8,0,1,1)
    uart.on('data')
  end
end
function recv_name(d)
  d = string.gsub(d, '\000', '')
  file.remove(d)
  file.open(d, 'w')
  uart.on('data', 130, recv_block, 0)
  uart.write(0, '\006')
end
function recv()
  uart.setup(0,9600,8,0,1,0)
  uart.on('data', '\000', recv_name, 0)
  uart.write(0, 'C')
end
```

Hoksmur, 2015-01-11:
@kmpm, good! Can you add example, how to use it?

Hoksmur, 2015-01-12:
If you do not mind, I would like to know the wishes of users.
[Vote on google docs](http://goo.gl/forms/4G75FiHHCh)

nodemcu, 2015-01-12:
go ahead, and thank you!

kmpm, 2015-04-07:
I am reopening this again.
I would really like to avoid having to upload and run these functions in lua every time it's needed.
A proper protocol implemented in for example the node api would be great.
Like node.recv() for example.

TerryE, 2015-11-02:
Sorry but we aren't going to implement this.

closed 2015-11-02 by TerryE

kmpm, 2015-11-06:
Ok, thanks anyway.
Any plans for something else to aid in uploading files?

TerryE, 2015-11-06:
@kmpm Peter, I use my own provisioning bootstrap, and do all this over wifi rather than having to plug my devices into a UART to load them .  Essentially I initialise the filesystem using [DiUS spiffsimg](/DiUS/spiffsimg) with this one small bootstrap file as its `init.lua` so that when you first turn on the esp8266, it picks up a DHCP address and uses a simple UDP ping pong protocol to download the full loader and initialise the SPIFFS file system from a hosted provisioning process (a Lua script which runs on an RPi connected to my router).  The script uses signed transfers and passes it's chipID, so the provisioning process can do device specific file system initialisation.   When I get around to it, I'll add this to the examples folder.