nodemcu/nodemcu-firmware #96: 0x091 printed

luismanson, 2015-01-11:
Not sure if this is a mistake on my code, but all text on my code has a 0x091 printed on it:
> dofile("main.lua")
```
Estados posibles de conexion
0 idle, 1 conectado, 2 err clave, 3 No AP, 4 fail, 5 Tengo IP
Detalles modulo:
STA MAC: \0x0918-FE-34-9C-D8-09
AP\0x091A-FE-34-9C-D8-09
chip: \0x0910278921
Memoria: \0x0911376
IP cliente: \0x09nil
MI AP ip\0x09192.168.4.1
```
the part of this code is:
```lua
print('Estados posibles de conexion')
print('0 idle, 1 conectado, 2 err clave, 3 No AP, 4 fail, 5 Tengo IP')
tmr.alarm(0, 1000, 1, function()
if wifi.sta.getip() == nil then
		print("Conectando al AP...")
		print ('Estado:', wifi.sta.status())
	else
		print('Conectado!')
		print('IP cliente: ', wifi.sta.getip())
		tmr.stop(0)
	end
end)
--detalles modulo
print('Detalles modulo:')
print('STA MAC: ',wifi.sta.getmac())
print('AP', wifi.ap.getmac())
print('chip: ',node.chipid())
print('Memoria: ',node.heap())
print('IP cliente: ', wifi.sta.getip())
print('MI AP ip', wifi.ap.getip())
```

Hoksmur, 2015-01-11:
I tested it few minutes ago - **\t** works corect, **\0xNN** doesnt works, **\0nn** - works, but not like octal, this way get *nn* (not *nnn* or *nnnn*) and interpret as decimal(!) code.

luismanson, 2015-01-11:
Thanks, but its not my intention to have those 0x09 nor tabs printed over serial in this specific code
I posted the issue to ask if it was a bug, If you see my code an its output im not requesting any \t to be shown  :(

kmpm, 2015-01-11:
As from http://www.lua.org/pil/2.4.html
> We can specify a character in a string also by its numeric value through the escape sequence \ddd, where ddd is a sequence of up to three *decimal* digits.

It doesn't take octal, only decimal.

nodemcu, 2015-01-12:
```
Estados posibles de conexion
0 idle, 1 conectado, 2 err clave, 3 No AP, 4 fail, 5 Tengo IP
Detalles modulo:
STA MAC:        18-FE-34-9F-BD-64
AP      1A-FE-34-9F-BD-64
chip:   10468708
Memoria:        19312
IP cliente:     nil
MI AP ip        192.168.4.1
> Conectando al AP...
Estado: 255
Conectando al AP...
Estado: 255
Conectando al AP...
Estado: 255
Conectando al AP...
```

here is my result. 
there is a 'tab'
maybe it's the serial terminal issue? I'm using teraterm.

Hoksmur, 2015-01-12:
Try test this fragment by Roger Meier's CoolTerm? The program can show output hexvalues.

nodemcu, 2015-01-12:
yes, '\t' is 0x09,
so find a term that can treat byte 0x09 as '\t' will solve this issue I think.

closed 2015-01-23 by nodemcu