nodemcu/nodemcu-firmware #100: flash_init_data_xx part in user_main() causes power cycle problem

nodemcu, 2015-01-12:
power cycles may failed to the prompt.
http://www.esp8266.com/viewtopic.php?f=18&t=1180

vowstar, 2015-01-16:
I've confirmed this issue. May we should do some special to restore init-data.

kubi57, 2015-01-18:
I just flashed 20150108 today and confirm that the ESP quite often doesn't power up correctly. I tried to connect RST with 10k to Vcc and 10uF to GND, but it didn't help. Indeed the previous firmware version worked well.

vowstar, 2015-01-20:
I try to use crc checksum to the sector before 0x10000 to make sure get the right state of boot. When ti is first boot, clear unused flash, burn initdata.

nodemcu, 2015-02-11:
guys is this still an issue?

kubi57, 2015-02-12:
I can make some tests this weekend, but it seems to be more stable now.

kubi57, 2015-02-16:
I wasn't able to reproduce failures while powering up. At least for me it's not an issue any more.

closed 2015-11-02 by TerryE