nodemcu/nodemcu-firmware #98: Webserver stops responding to requests via webpage

k-g, 2015-01-12:
I've written a small webserver app that always works the first time I connect, and then seemingly never again.

The goal is to turn an LED on and off via a webpage. I've been watching the power draw of the module and it appears to start off drawing around 70mA at 3.3V and then bouncing around after around a minute between 18 and 30 mA. 

So after a reset it boots, I can access the page, and then after a minute or so can no longer.

My nodeMcu version: NodeMcu 0.9.5 build 20150105  powered by Lua 5.1.4

I'm also using the LuaLoader app on windows for testing, but the behavior also happens when I access the page with my Android 4.4 phone.

Here's the code for reference:

```lua
 --pin #4 == GPIO2, pin #3 = GPIO 0
pin = 4
gpio.mode(pin, gpio.OUTPUT) 
print("Starting up...connecting")
wifi.setmode(wifi.STATION)
wifi.sta.config("MYSSID","MYPASSWORD")
wifi.sta.autoconnect(1)
tmr.delay(1000000)
for i=1,10 do
  if (wifi.sta.getip() ~= nil) then
    print("IP Address: "..wifi.sta.getip())
    break
  end
  tmr.delay(1000000)
  print(". ")
end

srv=net.createServer(net.TCP) 
srv:listen(80,function(conn) 
    conn:on("receive", function(client,request)
        local buf = "";

        --extract HTTP header fields
        local _, _, method, path, vars = string.find(request, "([A-Z]+) (.+)?(.+) HTTP");
        if(method == nil)then 
            _, _, method, path = string.find(request, "([A-Z]+) (.+) HTTP"); 
        end
        local _GET = {}

        --default pin setting
        local _pin_value = "OFF"

        --extract GET parameters from HTTP headers
        if (vars ~= nil)then 
            for k, v in string.gmatch(vars, "(%w+)=(%w+)&*") do 
                _GET[k] = v 
            end
            _pin_value = _GET.pin
        else

          --if no value was sent, read the present value for outputting as a status
          if(gpio.read(pin) == gpio.HIGH) then
            _pin_value = "ON"  
          else 
            _pin_value = "OFF"
          end 
        end

        --set the pin
        if(_GET.pin == "ON")then
              gpio.write(pin,gpio.HIGH);
        elseif(_GET.pin == "OFF")then
              gpio.write(pin,gpio.LOW);
        end

        ---build the webpage output form
        buf = buf.."<h1> Toggle the button</h1> <p>Pin is: [".._pin_value.."]. ";
        buf = buf.."<button onclick=\"location.reload()\">RELOAD</button></p>";
        buf = buf.."<p>Turn PIN2 <form src=\"/\"><button onclick=\"form.submit()\" name=\"pin\" value=\"ON\">ON</button></form>";
        buf = buf.."<form src=\"/\"><button onclick=\"form.submit()\" name=\"pin\" value=\"OFF\">OFF</button></form>";

        --send and close out everything
        client:send(buf);
        print(request);
    end)
    conn:on("sent",function(conn) conn:close() end)
end)
```

MarsTechHAN, 2015-01-30:
From my own test.It doesn't show any problem you said.You can try to updata to newest version to see weather the problem is been solved or not.
You can try to print the connection states to help you debug

ps. I strongly suggest to add those code.
```lua
 buf = "HTTP/1.1 200 OK \r\n\r\n"; ---HTTP Header
```
also, some of the features you use in your html code do not support by mobile version's explorer.

MarsTechHAN, 2015-01-30:
Also, you don't deal with the nil problem.
If I get http://.../?pin
The program will die and restart.

clixx-io, 2015-01-30:
What browser are you using?

Is your garbage collection in the right place?

One user was using Chrome and it was requesting the page multiple times and
dying.
That fixed it.

dvv, 2015-01-31:
You may try to respond with `HTTP/1.0 200 OK\r\n\r\n` -- notice 1.0 as version. This way the client should autoclose the connection, otherwise in absense of `Connection: close` header and 1.1 version the client will leave the connection open which may defeat your the logic.

nodemcu, 2015-02-11:
is this solved?

marcoskirsch, 2015-03-01:
I don't see this issue in nodemcu-httpserver (also in GitHub). It may be worth comparing the two or even using instead to do what you are trying to do.

Disclaimer: I wrote it
Disclaimer 2: It can still be improved upon.

mikewen, 2015-03-01:
Could be the problem of browser keep connection open. (Think your server is http 1.1)
And the second time, it just send the data without sending headers.
Your Lua code will out of memory and restart

I found using Lua find to parse the headers is not reliable, also Lua find does not support '\r\n\r\n'.

So when I was coding nodemcu-ide, I use javascript to post Lua codes to reduce memory usage:
https://github.com/NodeUSB/nodemcu-ide

TerryE, 2015-11-03:
See #719

closed 2015-11-03 by TerryE