nodemcu/nodemcu-firmware #99: Lua interpreter's limitation?

Hoksmur, 2015-01-12:
Hi, all.
I'm trying move part code to overlay and then use
```lua
myVal=loadfile("file.lua")( inparams)
```
But Lua crashed when file size more ~3 KiB. Ok, I understand why _loadfile_ crashed system: it must interpret all part and build function. But, I have the same problem when I'm trying use _dofile_. For explain some code:
```lua
-- myfile.lua, running "dofile('myfile.lua')
if not condition then
  return false
else
  -- is there a lot of code
  return true
end
```
So, _loadfile_ must use all code, but interpreter can drop all unparsed strings after 
```lua
return false
```
 My question:  is this feature Lua general engine or NodeMCU firmware?

Hoksmur, 2015-01-12:
Yes, engine interpret all file.  There is fragment from _lbaselib.c_:
```c
static int luaB_dofile (lua_State *L) {
const char *fname = luaL_optstring(L, 1, NULL);
int n = lua_gettop(L);
#if 0
if (luaL_loadfile(L, fname) != 0) lua_error(L);
#else
if (luaL_loadfsfile(L, fname) != 0) lua_error(L);
#endif
lua_call(L, 0, LUA_MULTRET);
return lua_gettop(L) - n;
}
```

nodemcu, 2015-01-12:
Yes, same as lua 5.1.4

Hoksmur, 2015-01-12:
I will use for my task _load_ or _loadstring_ and parse it part-by-part. 
Could you think of this functionality in the future? Optional, of course.  And I, too...
PS: My wants greater than my possibilities and time. :grimacing:

Hoksmur, 2015-01-12:
Is it possible, redirect strings from file to interpreter?

nodemcu, 2015-01-12:
file.readline() will read a line from a file.
```lua
f=loadstring("print('hello')") 
f()
```

nodemcu, 2015-01-12:
maybe, if we got time, or pull request.
we will modify lua engine to load and execute file from flash, line by line. to save ram.
for now, don't want to touch this lua engine part.

Hoksmur, 2015-01-12:
> file.readline() will read a line from a file.
Not correct. Your example will not work with function, if, while, repeat. When interpret gives 
_f=function()_ it waits _end_ . I will see _uart.on_ . Proposed way can help, if parsed source will write  in view of limitations. Not bad.

nodemcu, 2015-01-12:
lua engine can only do a complete(no error) chunk.
so maybe your file should be rewrite to fit this.
if .. then
else
--whole file
end
should be seperate into several small chunk:
```lua
function f1()
end
function f2()
end
if ... then
else
f1()
f2()
end
```

Hoksmur, 2015-01-12:
Should I rewrite interpreter? :smiley:
```lua
foo=function()
if cond then
-- blah-blah
elseif
 repeat
  if cond3 then
   while cond_n
   end
  end
 until cond2
end
end
```
How to check, how many strings need to read? :angry:
PS: Thanks a lot for trying to help me.

Hoksmur, 2015-01-12:
I will use _pcall_ _doadstring_ first time.
Small test:
```lua
str="a1, a2 = ...\
 local a3\
 a3=a1..a2\
 print('args: ', a1, a2)\
 return a3"

print( pcall( loadstring(str), "AA","DD" ) )
print("+++++++")
print( pcall( loadstring(str), "AA") )

str="a1, a2 = ...\
 local a3\
 a3=a1..a2\
 print('args: ', a1, a2)\
 return a3\
 if 1 then" --not finished

print("+++++++")
print( pcall( loadstring(str), "AA", "YY") )
```
>args: 	AA	DD
true	AADD
+++++++
false	[string "a1, a2 = ......"]:3: attempt to concatenate global 'a2' (a nil value)
+++++++
false	attempt to call a nil value

So, possible read string to buffer and try interpret it. If false - try buf=buf.."\r\n"..next .

closed 2015-02-11 by nodemcu