nodemcu/nodemcu-firmware #101: Binary File Access

3s1d, 2015-01-13:
Hi guys,
greate work!

Maybe I'm missing the big picture, but how do I access binary files? 

I'd like to build a webserver which offers compressed content (gzip) and some pictures. I need a way to store and retrieve binary data: e.g. file.write(array, length) or file.read(length)

Thanks
Juergen

kmpm, 2015-01-13:
I haven't tried this myself but
```lua
file.seek(startPos)
strData = file.read(length)
byteArray = pack(string.byte(strData))
```

Hoksmur, 2015-01-17:
What for do you need use compressed files? If you need send files more then 1460, maybe is better method to use _Accept-Ranges: bytes_ https://en.wikipedia.org/wiki/Hypertext_Transfer_Protocol 
This HTTP's feathures can help transmit big files from esp8266. I didn't try doit self.

3s1d, 2015-01-17:
Hi,
I'd like to transmit JPEG which can not be stored in a text format (unless I use BASE64 or something similar). And I'd like to user jQuery (or Zepto). Those java script files will not fit into the internal 512k flash side-by-side with the lua core unless they are compressed.
I hope I made my point clear. 

Unfortunately I had no time to look into this. But as soon as my LuaLoader is completed and I assembled my PCBs I come back to this issue.

Regards
Juergen

GeoNomad, 2015-01-27:
LuaLoader has binary file transfer.

http://benlo.com/esp8266/index.html#LuaLoader

https://github.com/GeoNomad/LuaLoader

3s1d, 2015-01-28:
Hi GeoNomand,

thanks for your response. I know about the LuaLoader and your site because I'm also into paragliding... However, I somehow overlooked this feature. Can you also read binary files?

Unfortunately, the Lualoader is Windows only. This would require a VM for me... Have you ever though about making it open source?

Regards,
Juergen

GeoNomad, 2015-01-31:
It will be open once I remove some extra stuff from the source. It was an adaptation of an old program with  a different purpose.

Right now there is no binary read, but that is a good idea. I might look at that.

Peter

nodemcu, 2015-02-11:
file.read()
file.write()
can do with binary.

closed 2015-02-11 by nodemcu

marcoskirsch, 2015-02-11:
I am planning to add binary file uploads to luatool.py. I already have some code. Let me know if you want to help.

dvv, 2015-02-11:
I'd like to have atomic file.rename to reliably upload files via write to
temp then rename pattern

kmpm, 2015-02-11:
http://github.com/kmpm/nodemcu-uploader might be of interest.
It uses a modified X-modem protocol for uploading.

dvv, 2015-02-11:
Good, thanks for pointers. But of my concern is loading modules by http which is a fixed protocol. So I would like to simply write what I `receive` via request body to a temp file then rename it to the target module name.
This way I wouldn't get stuck in inconsistent half-file state should mcu suddenly choked (which is so far very probable). See my point?

nodemcu, 2015-02-12:
file.rename() is in dev branch, will be merged to master and pre_build bin soon.