nodemcu/nodemcu-firmware #102: sending messages from one esp to another

michal2229, 2015-01-13:
Hi there. 
I am working on a project in which i need to communicate between two esp8266 modules. 
I am setting esp1 as a server AP and esp2 as a client. 
Then I am connecting client esp2 to a server esp1 AP via ssid and password.
Next I am creating connection on the client to server's IP (192.168.4.1) and trying to send a message, but I have no response. I tried many ways to communicate between these two modules but with no effect...
I have no problem with creating webserver trough AP, webserver trough router and sending to thingspeak every 15s on one device at the same time with 3 clients connected to it (PC and two smartphones) and refreshing page quite fast, so i doubt it is related to low heap or something like that... 
Client is connected to AP properly - print(wifi.sta.getip()) returns 192.168.4.2.

Could You help me figuring out how it should be done? Is it even possible to send message from one esp to another? 

Here is my test code:
```lua
wifi.setmode(wifi.STATIONAP)
cfg={}; 
cfg.ssid="ESP8266_" .. wifi.ap.getmac(); cfg.pwd="mypassword"; 
wifi.ap.config(cfg)
wifi.sta.config("ESP8266_1A-FE-34-9C-D7-70","mypassword") -- esp-03 - client
--wifi.sta.config("ESP8266_1A-FE-34-9E-0B-FE","mypassword") -- esp-01 - server
wifi.sta.connect() 

tmr.alarm(2, 10000, 1, function() -- I am sending message every 10s
	print("alarm 2...")
	print(wifi.ap.getip()) 
	print(wifi.sta.getip()) -- checking if connected succesfully
	conn2 = net.createConnection(net.TCP, 0)  
	conn2:on("receive", function(conn2, payload) 
		print(payload) -- nothing prints here, means there is no response... :(
		conn2:close()
		collectgarbage()
	end) 
	conn2:connect(80, "192.168.4.1") -- server/AP IP
	conn2:send("some message that I want to send to other (server) ESP")
	print("sent in alarm 2...")
end)

--server = net.createServer(net.TCP) 
--server:listen(80, function(conn) 
--	conn:on("receive", function(conn, receivedData)
--		print(receivedData)			
--		message = "message received, OK."
--		conn:send(message); message = nil -- here the response should be sended, but it does not even do "print(receivedData)", means there is no message received at all...
--	end) 
--	conn:on("sent", function(conn) 
--		conn:close() 
--		collectgarbage() 
--	end)
--end)
```

And here is my response trough uart from client:
*NodeMCU 0.9.5 build 20150108  powered by Lua 5.1.4
alarm 2...
192.168.4.1
192.168.4.2
sent in alarm 2...
alarm 2...
192.168.4.1
192.168.4.2
sent in alarm 2...*
and so on... 
Server, on the other hand, prints nothing...

Thanks in advance.

clixx-io, 2015-01-13:
Check the code here:
http://forum.makehackvoid.com/t/esp8266-workshop-january-21-attempting-useful-things/222/5

It looks like you are not waiting for the connection to open before sending
the data.

clixx-io, 2015-01-13:
Here's the code:

function transmit_msg()
   sk=net.createConnection(net.TCP, 0)
   sk:on("receive", function(sck, c) print(c) end )
   sk:on("connection", function(sck) sk:send("ButtonPress\r\n") end )
   sk:connect(9000,"192.168.0.6")
   tmr.delay(1000000)
   print("Sent")
   end

On Tue, Jan 13, 2015 at 6:41 PM, David Lyon <david.lyon.preisshare@gmail.com

michal2229, 2015-01-13:
Thank You, I will give it a try.

michal2229, 2015-01-14:
I figured out that main cause of problem was the wifi mode of client ESP. 
It does work with STATION, but doesn't work with STATIONAP mode at all. I am guesing that it is caused by the same acces point IP of two ESP's. Am I right?
Suprisingly erasing tmr.delay() line didn't affect sending negatively. I have erased it because it was causing problems with code uploading via FTDI. 
Sending 'on("connection"...' works for me very well :)

The following code works:
```lua
-- CODE FOR CLIENT ESP
wifi.setmode(wifi.STATION) -- must be STATION here, STATIONAP doesn't send messages properly. Maybe it is trying to send messages to itself? Or IP conflict?
wifi.sta.config("ESP8266_1A-FE-34-9E-0B-FE","mypassword") -- connecting to server
wifi.sta.connect() 

tmr.alarm(2, 250, 1, function()
    if (wifi.sta.getip() ~= nil) then 		
		sk = net.createConnection(net.TCP, 0)  
		sk:on("receive", function(sk, receivedData)
			print("received: " .. receivedData) 
			sk:close()
		end) 
		sk:on("connection", function(sk) sk:send("Hello, do you read me?") end )
		
		sk:connect(80, "192.168.4.1") -- server ESP IP
		print("Message sent.")
	else
		print("no connection...")
	end
    collectgarbage() 
end)
```

```lua
-- CODE FOR SERVER ESP
wifi.setmode(wifi.STATIONAP)
cfg={}; cfg.ssid="ESP8266_" .. wifi.ap.getmac(); cfg.pwd="mypassword"; wifi.ap.config(cfg)

server = net.createServer(net.TCP) 
server:listen(80, function(conn)
	conn:on("receive", function(conn, receivedData) -- zmienic conn w funkcji na cos innego
		print("received: " .. receivedData)         
		message = "Hi, I read You."
		conn:send(message); message = nil
	end) 
	conn:on("sent", function(conn) 
      collectgarbage() 
	end)
end)
```

Thanks for Your help.

closed 2015-02-11 by nodemcu

ashwinipatil123, 2015-05-06:
I am doing the same thing I can establish communication between 2 esps but I want to upload I file from esp client to server.
How to upload a file in nodemcu firmware using lua

TerryE, 2015-05-06:
Again, this is a developer Q and not and issue with the firmware.  Please close this issue and raise it on the [esp8266.com nodeMCU forum] (http://www.esp8266.com/viewforum.php?f=17) or StackOverflow