nodemcu/nodemcu-firmware #103: Latest Download and Eclipse

scargill, 2015-01-13:
Downloaded latest updates today. Renamed to nodemcu-firmware. Imported into Windows (/espressif/examples) just as any other project. Imported.  Ran CLEAN.

Error: Program "make" not found in PATH.

Still something wrong.

nodemcu, 2015-01-13:
add
C:\MinGW\bin;C:\MinGW\msys\1.0\bin; C:\Python27;to system environment PATH

scargill, 2015-01-13:
Thanks. Did that – compilation worked right through to the end then..

 

 

xtensa-lx106-elf-gcc  -L../lib -nostdlib -T../ld/eagle.app.v6.ld -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--start-group -lc -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lmain -ljson user/.output/eagle/debug/lib/libuser.a driver/.output/eagle/debug/lib/libdriver.a lwip/.output/eagle/debug/lib/liblwip.a json/.output/eagle/debug/lib/libjson.a ssl/.output/eagle/debug/lib/libssl.a upgrade/.output/eagle/debug/lib/libupgrade.a platform/.output/eagle/debug/lib/libplatform.a libc/.output/eagle/debug/lib/liblibc.a lua/.output/eagle/debug/lib/liblua.a smart/.output/eagle/debug/lib/smart.a wofs/.output/eagle/debug/lib/wofs.a spiffs/.output/eagle/debug/lib/spiffs.a modules/.output/eagle/debug/lib/libmodules.a -Wl,--end-group -o .output/eagle/debug/image/eagle.app.v6.out 

../tools/esptool.py elf2image .output/eagle/debug/image/eagle.app.v6.out -o ..\\bin\\

make[1]: ../tools/esptool.py: Command not found

make[1]: *** [.output/eagle/debug/bin/eagle.app.v6.bin] Error 127

make[1]: Leaving directory `/c/Espressif/examples/nodemcu-firmware/app'

make: *** [.subdirs] Error 2

 

11:55:37 Build Finished (took 1m:866ms)

nodemcu, 2015-01-14:
did you install python 2.7
and pyserial?

kmpm, 2015-01-14:
Could you, just as a test,  change the Makefile.
Where it says

__ESPTOOL = ../tools/esptool.py__ to

__ESPTOOL = python ../tools/esptool.py__

scargill, 2015-01-15:
yes

scargill, 2015-01-15:
Hi

 

In Windows – what you’ve supplied doesn’t look valid  - even ignoring the underscores.

 

I even tried putting the full path in as used by others…

 

One of these would normally work.

 

ESPTOOL = c:\Espressif\utils\esptool.py

ESPTOOL = c:\Espressif\utils\esptool.exe

 

And yes python and pyserial are in there. I don’t think that’s where the problem is – surely that’s for blowing chips? This is when running the ALL command..

 

 

c:\Espressif\utils\esptool.py elf2image .output/eagle/debug/image/eagle.app.v6.out -o ..\\bin\\

/usr/bin/env: python: No such file or directory

make[1]: *** [.output/eagle/debug/bin/eagle.app.v6.bin] Error 127

make[1]: Leaving directory `/c/Espressif/examples/nodemcu-firmware/app'

make: *** [.subdirs] Error 2

nodemcu, 2015-01-18:
add some more path to env in eclipse project.

closed 2015-01-18 by nodemcu