nodemcu/nodemcu-firmware #104: IP Forwarding not set, can't be done without recompile?

tz1, 2015-01-13:
I need the ESP8266 to be able to forward packets between two connected devices.  Currently it does not, i.e. I can have my PC and my tablet and the two while connected to the ESP-as-AP can't see or communicate with each other.

In some *opt.h is an IP_FORWARD that is set to zero, so is turned off.
Would it be possible to either turn this on by default or create some net option to allow this which I assume will fix the problem?

vowstar, 2015-01-20:
We checked the IP_FOWARD option in lwip, and it is default off. We will test what happend when it open. It may recompile lwip library can solve this problem.

zendes, 2015-01-20:
You also need to enable IP_FORWARD_ALLOW_TX_ON_RX_NETIF, since the ESP8266 would be sending the packet back out the same network interface.

eku, 2015-10-20:
How is the routing between the networks configured?

kaboom748, 2015-12-25:
COMMENT 2 SECTIONS

 /* Do not forward packets onto the same network interface on which
   * they arrived. */
 // if (netif == inp) {
//    LWIP_DEBUGF(IP_DEBUG, ("ip_forward: not bouncing packets back on incoming interface.\n"));
 //   goto return_noroute;
//  }

  /* decrement TTL */
  IPH_TTL_SET(iphdr, IPH_TTL(iphdr) - 1);
  /* send ICMP if TTL == 0 */
  if (IPH_TTL(iphdr) == 0) {
    snmp_inc_ipinhdrerrors();
#if LWIP_ICMP
    /* Don't send ICMP messages in response to ICMP messages */
    if (IPH_PROTO(iphdr) != IP_PROTO_ICMP) {
     // icmp_time_exceeded(p, ICMP_TE_TTL);
    }
#endif /* LWIP_ICMP */

kaboom748, 2015-12-26:
1) xtensa-lx106-elf-ar x liblwip.a [EXTRACT LIB]
Compile ip.c and icmp.c ---> ip.o and icmp.o
3) xtensa-lx106-elf-ar cr liblwip.a file.o file2.o file3.o................. [REPACK MODIFIED LIB]

Be sure --> include/lwipopts.h and include/lwip/opt.h =====> IP_FORWARD 1

kaboom748, 2015-12-26:
If Unresolved symbols..............
xtensa-lx106-elf-nm liblwip.a ..............
T ==> LIB OK
U ==> LIB NOT OK
t ===> just remove the "STATIC" before the function for use it

kaboom748, 2015-12-26:
The routing with the AccessPoint and Station work perfect.

TESTED........

kaboom748, 2015-12-26:
#!/bin/sh

xtensa-lx106-elf-ar cr liblwip.a api_lib.o     espconn_mdns.o  inet.o     netbuf.o    sntp.o      tcp_out.o \
api_msg.o     espconn.o       init.o     netdb.o     sockets.o   timers.o \
autoip.o      espconn_tcp.o   ip_addr.o  netifapi.o  stats.o     udp.o \
def.o         espconn_udp.o   ip_frag.o  netif.o     sys_arch.o \
dhcp.o        etharp.o        ip.o       netio.o     sys.o \
dhcpserver.o  icmp.o          mdns.o     pbuf.o      tcp_in.o \
dns.o         igmp.o          mem.o      ping.o      tcpip.o \
err.o         inet_chksum.o   memp.o     raw.o       tcp.o

kaboom748, 2015-12-26:
#!/bin/sh

export CFOLDER="-I/root/LWIP/LWIP/include -I/root/esp-open-sdk/sdk/include"
export CFLAGS="-Os -g -O2 -Wpointer-arith -Wundef -Werror -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals  -D__ets__ -DICACHE_FLASH"

export LFOLDER="-L/root/esp-open-sdk/sdk/lib"
export LDFLAGS="-nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static"
export LIBS="-lc -lgcc -lhal -lpp -lphy -lnet80211 -llwip -lwpa -lmain"

export LIBPATCH="-T/root/esp-open-sdk/sdk/ld/eagle.app.v6.ld"

export FICHIERS_C="ip.c"

xtensa-lx106-elf-gcc $CFOLDER $CFLAGS -c $FICHIERS_C -o ip.o

TerryE, 2015-12-26:
@kaboom748, can you please keep to the standard Github conventions on posting to issues.  Use markdown formatting.  Raise an issue explaining what you are doing / asking and why.  Develop a patch and issue a pull request.   Don't expect us to try to interpret what seems gibberish.  

If you can't conform to this request, I will delete your posts.

kaboom748, 2015-12-26:
And Finally this is the file patched with the IP_FORWARD activated

kaboom748, 2015-12-26:
I cannot post the liblwip.a file............. Unfortunately, we don’t support that file type.  Try again with a PNG, GIF, JPG, DOCX, PPTX, XLSX, TXT, PDF, or ZIP. 

And if I Zip the file... same error code

TerryE, 2015-12-26:
Thanks, I will tidy up your previous posts, but please stick to the Github practices for posting.  Marcel or I can provide a translation as I have just done, but we will reply in English so that other's can understand the dialogue.  Google translate is your friend.

devyte, 2016-03-17:
I just tested the code changes mentioned here in the dev branch:
*#define IP_FORWARD 1
*comment out the if block
*comment out the exceed thing

It doesn't seem to work. I can ping both netifs on the ESP, from both sides, but I can't ping through the ESP.
E,g;
From a station client connected to the ESP I can ping both the ESP ap interface and the client interface, and from my home router where the ESP is connected I can ping both the ESP's sta and ap interfaces, but I can't ping from the client through the ESP to the router.
laptop ---> (ap)ESP(sta) ---> router <---- other laptop

If someone knows anything else to try, I'm willing to test it out.

devyte, 2016-03-20:
Ok, I have to retract my previous post: this does in fact work in the latest dev branch (1.5.1).
My test setup:
router() <--> (sta)ESP(ap) <--> laptop
router: 192.168.0.1 (static)
sta IP: 192.168.0.111 (dhcp - router)
ap IP: 192.168.111.1 (static)
laptop IP: 192.168.111.100 (dhcp - ESP)

Steps I followed:
* app/include/lwipopts.h: changed #define IP_FORWARD from 0 to 1
* app/include/lwip/opt.h: changed #define IP_FORWARD from 0 to 1
* app/lwip/core/ipv4/ip.c: commented out the following
```
// if (netif == inp) {
 //    LWIP_DEBUGF(IP_DEBUG, ("ip_forward: not bouncing packets back on incoming interface.\n"));
 //   goto return_noroute;
//  }
```
* added a static route to my wifi router: destination network: 192.168.111.0 and gateway address: 192.168.0.111
* added reserved address to my ESP's mac address, so that is always gets the same IP assigned.
* loaded the following DNS forwarding script in the ESP:
```
 cu=net.createConnection(net.UDP,0)  
 cu:on("receive",function(cu,c)   
   print("Got a reply")  
   s:send(c)  
   end)  
 s=net.createServer(net.UDP)  
 s:on("receive",function(s,d)  
   print ("Got a request!")  
   cu:connect(53,net.dns.getdnsserver())  
   cu:send(d)   
   end)  
 s:listen(53) 
```

After the above, I was able to browse on my laptop, and even stream netflix through the ESP, which is not bad at all. In essence, the ESP acts like a wifi range extender or hotspot.
I'll admit I didn't test thoroughly, tho...

devyte, 2016-03-20:
Forgot to mention, credit is not mine, but [Andreas Reischle](http://www.areresearch.net/2015/10/using-esp8266-as-wifi-range-extender.html). Only difference from his solution is that I didn't hardcode an external dns server in the dns forwarder, but instead I get the one currently set via dhcp (recent addition to dev branch).

devyte, 2016-03-20:
@TerryE @pjsg The lua dns forwarder above apparently can't be shutdown, it seems to leak in the sense that the connection and socket don't release resources when closed:
```
s:close()
cu:close()
s,cu = nil
```
Even after the above, the registry shows the callbacks, and the forwarder keeps working. 
I suspect this is because of the crossed s/cu references. If that's the case, then I think we may need a way to unregister event callbacks, e.g.: allow something like this:
`s:on("receive", nil) -- if a callback was set, it gets cleared`

Would you agree? If so, should I open a new issue?

BTW, the above is UDP, and I know there is some discussion going on about the UDP use model, but the same principle applies to a TCP forwarder.

jmattsson, 2016-04-15:
Terry has started the Herculean task of cleaning up the net module, stay tuned for more updates on that front.

kiki67100, 2016-06-28:
@kaboom748 pourrais tu me dire les etapes que tu as suivi pour arriver à ça ? J'ai compiler le dossier src de lwip mais cela ne change rien.

closed 2019-06-08 by TerryE