nodemcu/nodemcu-firmware #105: Generalize HTTP parser

dvv, 2015-01-16:
Hi!

webap_toggle_pin.lua example provides some clues to how to write a web server. Wonder if it HTTP parser can be introduced which does low-level job of extracting method, path, (version?), headers, body. Can be done in form of `:on` callbacks, say, `c:on("method", function(method) ... end)` etc.
Though it can be done in Lua, I believe, interning of strings plays contra here, and such a parser is essentially all about strings.

TIA,
--Vladimir

NeiroNx, 2015-01-16:
it is only limits by memory. I am try full versions of parser, but i am need reduse it functionality because app has crash on low memory. Memory may be needs to other tasks.

dvv, 2015-01-17:
Well, no need in _full_ parser -- just such calling callbacks for important HTTP parts needed. Method, and URL (as is) and a callback for "body is started" already would be nice.
May be in form of a single callback `:on("req", function(conn, method, url) ... end)`.

Hoksmur, 2015-01-17:
I'm using overlyas and user's function callback when answer is generated. and I agree with @NeiroNx - Lua spend memory fast! My project is so raw, but you can see methods - https://github.com/Hoksmur/uLuaNode

dvv, 2015-01-17:
I don't see why you are so careful of memory. I propose to do stringy stuff in C and fire some additional callbacks. I do _not_ propose to _collect_ all the headers and the body before heading them off to handler.

Plus, I believe you don't do anything with POST and other "bodyful" requests which may arrive in chunks (1460 octets at most).

And I agree that for `GET /pin?k=1&v=0`-alike requests when nothing is to care of expect URL the current set of instruments is fine.

So even if having "http" module inside the firmware is not feasible, an example of _proper_ HTTP parser will be welcome

Hoksmur, 2015-01-17:
Sorry, I didn't catch the main idea at first.

dvv, 2015-01-17:
That's what I have so far, but it sometimes fails down to reset. Probably a watchdog triggered, don't know how to debug, yet.

```lua
srv = net.createServer(net.TCP, 15)
srv:listen(80, function(c)
  -- HTTP parser
  local buf = ""
  local method, url
  local ondisconnect = function(conn)
    print("FIN")
    collectgarbage("collect")
  end
  local cnt_len = 0
  local onheader = function(conn, k, v)
    -- TODO: look for Content-Type: header
    -- to help parse body
    print("H", k, v)
    -- parse content length to know body length
    if k == "content-length" then
      cnt_len = tonumber(v)
    end
    if k == "expect" and v == "100-continue" then
      conn:send("HTTP/1.1 100 Continue\r\n")
    end
  end
  -- body processed, we may ultimately reply here
  local onend = function(conn)
    conn:send("HTTP/1.1 200 OK\r\n\r\n")
    print("CLOSE")
    conn:close()
  end
  local body_len = 0
  local ondata = function(conn, chunk)
    -- NB: once length of seen chunks equals Content-Length:
    --   onend(conn) is called
    body_len = body_len + #chunk
    print("B", #chunk, body_len, cnt_len, node.heap())
    if body_len >= cnt_len then
      onend(conn)
    end
  end
  local onrequest = function(conn, method, url)
    print("R", method, url)
  end
  local onreceive = function(conn, chunk)
    print("IN", #chunk)
    -- merge chunks in buffer
    if buf then
      buf = buf .. chunk
    else
      buf = chunk
    end
    -- consume buffer line by line
    while #buf > 0 do
      -- extract line
      local e = buf:find("\r\n", 1, true)
      if not e then break end
      local line = buf:sub(1, e - 1)
      buf = buf:sub(e + 2)
      -- method, url?
      if not method then
        local i
        -- NB: just version 1.1 assumed
        _, i, method, url = line:find("^([A-Z]+) (.-) HTTP/1.1$")
      -- header line?
      elseif #line > 0 then
        -- parse header
        local _, _, k, v = line:find("^([%w-]+):%s*(.+)")
        -- header seems ok?
        if k then
          k = k:lower()
          onheader(conn, k, v)
        end
      -- headers end
      else
        -- request callback
        onrequest(conn, method, url)
        -- NB: we feed the rest of the buffer as starting chunk of body
        ondata(conn, buf)
        -- buffer no longer needed
        buf = nil
        -- NB: we explicitly reassign receive handler so that
        --   next received chunks go directly to body handler
        conn:on("receive", ondata)
        -- parser done
        break
      end
    end
  end
  c:on("receive", onreceive)
  c:on("disconnection", ondisconnect)
end)
```

clixx-io, 2015-01-17:
Is there some way to debug these on the pc?

dvv, 2015-01-17:
Well, I feed it in PC Lua with something like
```lua
onreceive("POST /up")
onreceive("load?baz")
onreceive("=qux HTT")
onreceive("P/1.1\r\n")
onreceive("User-Age")
onreceive("nt: curl")
onreceive("/7.30.0\r\n")
onreceive("Host: 19")
onreceive("2.168.1.")
onreceive("6\r\nAcc")
onreceive("ept: */*\r\n")
onreceive("\r\nfoo=")
onreceive("bar")
```

etc.

However, real mcu fails to sink longer POST requests (5kB of data) with _no info_ printed via `print(...)` statements...

iMarvin1337, 2015-01-18:
I found a library for parsing URI queries in LUA.
Maybe this could be implemented in a release.
https://github.com/golgote/neturl

dvv, 2015-01-20:
The issue partly addressed here https://github.com/nodemcu/nodemcu-firmware/commit/b6a403010294af4b5f011e7ca6782ace0e28a83e

mimil, 2015-05-20:
+1

Http parser is also needed for http client calls (my current need), much of the code I see is for http server calls.

marc-gist, 2015-08-24:
I too am looking for an easy way to parse out (I don't care about the headers returned in an http client call) just the body of a response from a web server.
Any help/pointers to some completed code would be great. thanks.

TerryE, 2015-11-02:
Sorry but we're not going to implement this unless someone offers an implementation.

closed 2015-11-02 by TerryE