nodemcu/nodemcu-firmware #106: conn:send(table)

dvv, 2015-01-16:
Hi!

Please, consider making connection sender method accept as well a table of strings. This would allow for getting rid of costly string concatentions.

TIA,
--Vladimir

vowstar, 2015-01-20:
Good idea

dvv, 2015-01-20:
I looked at net_send code. Even started to hack but realised there should be not a cycle through table, but a recursive call like:
```lua
function conn:send(x, cb)
  if #x == 0 then
    cb(self)
  else
    self:send(x[1], function()
      self:send(x_without_1_element, cb)
    end)
  end
end
```
only in C which is out of my skill...

TerryE, 2015-11-02:
`conn:send(table.concat(table))` works for totals less than the packet size but we have rto work withing the SDK. This is an enhancement that isn;t going to be implemented by any of the current contributors.

closed 2015-11-08 by TerryE