nodemcu/nodemcu-firmware #110: TCP connection does not appear to use random port for SRC port?

xlfe, 2015-01-17:
I have my ESP module setup to send temperature data to the server at regular intervals, and as part of the debugging process I've been examining the TCP packets it sends/receives.

Not sure why but it is consistently using a SRC port of 4097 when connecting to a HTTP server on port 80

[image]

xlfe, 2015-01-17:
Just to confirm - different devices use different ports - ie one is always using SRC port 4097, another is always using SRC port 1038...

xlfe, 2015-01-18:
Ok.. after further analysis it looks to be increasing. So i have two modules where the SRC port (is increasing by one each time) in the 1xxx range and a third module which is stuck on 4097 for source port.

nodemcu, 2015-01-18:
please try latest build 20150118

closed 2015-01-18 by nodemcu

xlfe, 2015-01-19:
Thanks @nodemcu - that fixed it. I just wanted to say that I really appreciate all the time and effort you and your team are putting into NodeMCU - very much appreciated!

Cheers