nodemcu/nodemcu-firmware #112: Multiple tcp servers?

clienthax, 2015-01-18:
Is it possible to have more than one tcp server,
i am hijacking the serial io for gpio and was going to have a http and telnet lua console
but i get a message saying i can only have one tcp server

clixx-io, 2015-01-18:
how do you hijack the serial to use it as GPIO?

clienthax, 2015-01-19:
@clixx-io 
gpio.mode(9,gpio.OUTPUT) --rx
gpio.mode(10,gpio.OUTPUT) --tx linked to led, might cause issues

  lighton=0
  tmr.alarm(1,1000,1,function()
    if lighton==0 then 
      lighton=1 
	  gpio.write(9,gpio.HIGH)
	  gpio.write(10,gpio.HIGH)	 
    else 
      lighton=0 
	  gpio.write(9,gpio.LOW)
	  gpio.write(10,gpio.LOW)
    end 
  end)

clixx-io, 2015-01-19:
@clienthax: Thank you.

vowstar, 2015-01-20:
Current sdk v0.9.5 limited us use only one tcp server. But the lwip is open source, change the espconn api can get more tcp server.

evgenitznikolov, 2015-03-17:
I want to achieve the same. Any chance to enable more than one TCP server? Actually you can let ESP listen to two ports (kind of bug I guess) but I couldn't find a way in the event procedure to identify the port on which the data has arrived. Any ideas?

TerryE, 2015-11-02:
> the espconn api can get more tcp server.

We buld nodeMCU firmware on top of the SDK. Sorry guys.

closed 2015-11-02 by TerryE

grahamehorner, 2015-11-03:
This definitely prevents us from running http and https at the same time, given the limit of one tcp server, personal I'd always opt for https over http any day, (when it stable) but worth noting as a limitation of the current sdk.