nodemcu/nodemcu-firmware #113: Can only write to file 217 times before filesystem failure

xlfe, 2015-01-18:
It seems that after a writing to a file 217 times the filesystem no longer operates.

```lua
count = 0 
while 1 do
    print(file.open("test.test","w+"))
    file.write(count)
    file.close()
    print("Count: "..count)
    count = count + 1 
end
```

What happens :
```lua
true
Count: 216
true
Count: 217
�R�Y
    ���B
NodeMCU 0.9.5 build 20150118  powered by Lua 5.1.4
lua: cannot open init.lua
```

format the filesystem fixes it
```lua
file.format()
```
but it occurs again. Have tried on two different modules, both stop at 217.

raz123, 2015-01-19:
Wow. This is a serious issue. My application also requires periodic logging to a file.
Have you tried with older firmwares?

xlfe, 2015-01-19:
@raz123 no only on firmware 20150118

raz123, 2015-01-19:
@xlfe, tested on 20150108. No bug here. ESP8266-01.

xlfe, 2015-01-19:
Thanks @raz123 - can you confirm you have the bug on 20150118?

Thanks

vowstar, 2015-01-20:
Could you tell me what hardware do you use? I need more infomation about the flash size and it will help us test and fix it.

xlfe, 2015-01-20:
ESP-03 - 4MBit / 512kb flash

raz123, 2015-01-20:
@xlfe I tried with 20150118, again using an ESP-01, and no bug here. Easily made it to over a thousand with the code you provided.

```
Count: 1323
true
Count: 1324
true
Count: 1325
```

xlfe, 2015-01-20:
Im having difficulty reproducing the bug on stock firmware.. sorry! let me investigate further

xlfe, 2015-01-21:
Build nodemcu with [BMP280](https://github.com/xlfe/nodemcu-firmware/tree/xlfe) module
Crashes around ~200 writes
0x00000.bin 46,144 bytes
0x10000.bin 312,776 bytes 0x4C5C8

Comment out LUA_USE_MODULES_BMP =~stock build
No problem apparent (>13,000 writes)
0x10000.bin 45,024 bytes 
0x10000.bin 310,904 bytes 0x4BE78

Comment out LUA_USE_MODULES_BIT
No problem apparent (>1000 writes)
0x00000.bin 44,928 bytes
0x10000.bin 309,884 bytes 0x4BA7C

Comment out LUA_USE_MODULES_OW
Crashes around ~200 writes
0x00000.bin 44,736 bytes
0x10000.bin 306,116 bytes 0x4ABC4

**My guess is that the bug is caused when the 0x10000.bin is larger or smaller than 0x4B?**

I can get my custom build with BMP280 module to work without the bug if the size of 0x10000.bin is 308160 bytes ie 0x4B3C0

You should be able to reproduce it on stock firmware by disabling modules or other things that make the 0x10000.bin size greater or smaller than 0x4bXX

xlfe, 2015-01-26:
Just tried on the latest pre_build binary (no modifications):

```lua
Count: 171
true
Count: 172
�R
  �Y�

NodeMCU 0.9.5 build 20150127  powered by Lua 5.1.4
lua: cannot open init.lua```

However, If I modify the build so it ends on the 0x4B boundary:

```bash
-rw-rw-r--  1 build build   43280 Jan 27 10:15 0x00000.bin
-rw-rw-r--  1 build build  309324 Jan 27 10:15 0x10000.bin```

Then it works as expected...

vowstar, 2015-01-27:
I have confirmed this bug, and finding the reason.

vowstar, 2015-01-27:
I tested it, and I found if file system's length (use 0x7C000-STARTADDR) is 0x20000 or 0x1C000 or 0x24000 is OK. That mean's file system length must divided by 0x4000.
Now iromtext begin at 0x10000, file system begin at STARTADDR = (0x10000 + lengthof(iromtext) + 0x1000) & (0xFFFFF000), file system's length may not divide by 0x4000.
Align file system's STARTADDR to 0x4000 can fix this bug, but we will test again to find a better solution.

nodemcu, 2015-01-27:
quick fix this by align the start address to 0x4000. pls try.

xlfe, 2015-01-27:
Thanks - will give it a shot

closed 2015-02-11 by nodemcu

xlfe, 2015-02-18:
Thanks this has fixed the bug!

xlfe, 2015-02-18:
For anyone else interested, python code to calculate the offset to write the filesystem to:

```python
#size of 0x10000.bin
sz = 285312
adj = sz/0x4000*0x4000+0x4000
print 'Write to {:02x}'.format(adj+0x10000)
```