nodemcu/nodemcu-firmware #117: File over 18999 bytes causes crashes

raz123, 2015-01-19:
Creation of a file of 18999 bytes is not an issue. Everything works as expected.

Creation of a file of 19000 bytes or more causes the whole thing to crash and misbehave. The only recourse to recovery, besides flashing again, is issuing a file.format() command.

This is problematic. I believe that the ESP8266 with large flashes (4 Mb) were introduced to allow users to play with "large" files (at least, larger than 19 kbytes).

vowstar, 2015-01-20:
It may single file bug of filesystem, we are checking why one file can't 19000 bytes.

nodemcu, 2015-02-11:
could you please try to build and test the "dev" branch?
https://github.com/nodemcu/nodemcu-firmware/tree/dev
I think this is solved, I update file system. but need some more test.

nodemcu, 2015-02-11:
my test result
```lua
> file.open("large.lua","w+")
> for i=1,19100 do file.write("a") end
> file.close()
> print(file.list()["large.lua"])
19100
>
```

raz123, 2015-02-16:
Thanks, seems to be corrected, although there's a loss in overall space.

Max writes:

20150212: 1*67056 = 67,056 bytes.
20150107: (4*18999)+11684+2540 = 90,220 bytes.

closed 2015-03-09 by nodemcu