nodemcu/nodemcu-firmware #121: mqtt branch - Unable to initialize mqtt client

MrBalonio, 2015-01-21:
Hello, 

I am running the lastest bin and after connecting to wireless I am unable to initialize the MQTT client

Here's what I get

> m = mqtt.Client("clientid", 120, "user", "password")
stdin:1: attempt to call field 'Client' (a nil value)

I also tried 
 m=mqtt.Client()
stdin:1: attempt to call field 'Client' (a nil value)

robb-aker, 2015-01-21:
Hey,

I hit the same problem. The old method for creation still works.

rob

nodemcu, 2015-01-21:
in the branch mqtt， you have to compile the code yourself。

tuanpmt, 2015-01-21:
Binary file can be downloaded here: http://www.esp8266.com/viewtopic.php?f=19&t=1278

MrBalonio, 2015-01-21:
@tuanpmt  thanks. I will try the new binary with the new example

nodemcu, 2015-01-23:
mqtt branch merged to master, pre_build bin included.

closed 2015-01-23 by nodemcu

robb-aker, 2015-01-23:
Works lovely. Thanks for the great work guys!

MrBalonio, 2015-01-24:
Thanks it works great