nodemcu/nodemcu-firmware #123: First call to wifi.sta.getap after mode change passes a nil table

hamishcunningham, 2015-01-21:
I wonder if this behaviour is correct? After doing wifi.setmode I call wifi.sta.getap twice; the first time, the callback function is passed a nil parameter. (Code and output below.) 

Should I code to expect this, or is it a bug?

Thanks! Hamish

    -- init.lua
    wifi.setmode(wifi.STATIONAP)
    function listap(t)
      if t then 
        for k,v in pairs(t) do print(k.." : "..v) end
      else
        print("getap feeding nil table to callback function")
      end
    end
    wifi.sta.getap(listap) -- prints message about nil
    wifi.sta.getap(listap) -- prints list of available APs

    NodeMCU 0.9.5 build 20150118  powered by Lua 5.1.4
    getap feeding nil table to callback function
    > 
    TALKTALK-4A6654 : 4,-90,78:54:2e:4a:66:54,2
    O2wireless892434 : 1,-73,00:1d:68:6c:1e:75,1
    NETGEAR66 : 3,-90,20:4e:7f:e3:03:12,6
    BTHub4-HPSZ : 3,-95,94:fb:b2:7e:8e:4c,11

(I'm running the latest nodemcu from github, rebuilding with esp-open-sdk, flashing with esptool.py, and using ESPlorer 2.06 to connect to the ESP, on Ubuntu 14.04.)

TheFoxy, 2015-01-21:
NodeMCU 0.9.5 build 20150118  powered by Lua 5.1.4
same problem here.
if you write 
...
wifi.sta.getap(listap) -- prints message about nil
wifi.sta.getap(listap) -- prints list of available APs
wifi.sta.getap(listap) -- prints list of available APs
wifi.sta.getap(listap) -- prints list of available APs
wifi.sta.getap(listap) -- prints list of available APs
always the last is valid.. other are nil
maybe a timing problem  ;)

macfisherman, 2015-01-25:
I don't think it is a timing problem.

````lua
wifi.setmode(wifi.STATION)
scan = true

function listap(t)
  if type(t) ~= "table" then
    print("t is " .. type(t))
    scan = true
    return
  end
  print("scanning")
  for k,v in pairs(t) do
    print(k.." : "..v)
  end
  scan = true
end

tmr.alarm(0, 1000, 1, function ()
     if scan then
       scan = false
       wifi.sta.getap(listap)
     else
       print("waiting...")
     end
   end
)
````
I get crashes no matter what.
````
> waiting...
t is nil
H!�ȩ��FJH9�D�!��@H��H�
NodeMCU 0.9.5 build 20150118  powered by Lua 5.1.4
lua: cannot open init.lua
````

It did work for a bit, but then the system froze. I have a nodemcu-dev board. The power to it is strictly from the USB port. I think it is more likely a overflow of some sort. I have lots of reachable APs in certain sections of my house.

vowstar, 2015-01-25:
Could you help us test this code?
````lua
wifi.setmode(wifi.STATION)
scan = true

function listap(t)
  tmr.wdclr()
  if type(t) ~= "table" then
    print("t is " .. type(t))
    scan = true
    return
  end
  print("scanning")
  for k,v in pairs(t) do
    print(k.." : "..v)
  end
  scan = true
end

tmr.alarm(0, 1000, 1, function ()
     if scan then
       scan = false
       wifi.sta.getap(listap)
     else
       print("waiting...")
     end
   end
)
````

hamishcunningham, 2015-01-27:
hi -- here's some output from your code. 
btw, I've also noticed that the callback function is sometimes called twice -- perhaps related?
cheers
h

> t is nil
waiting...
waiting...
scanning
SKY907DA : 3,-93,7c:4c:a5:e3:49:89,6
O2wireless892434 : 1,-89,00:1d:68:6c:1e:75,1
pibrush : 3,-50,44:94:fc:74:27:a4,11
t is nil
waiting...
waiting...
scanning
SKY907DA : 3,-90,7c:4c:a5:e3:49:89,6
O2wireless892434 : 1,-89,00:1d:68:6c:1e:75,1
pibrush : 3,-52,44:94:fc:74:27:a4,11
t is nil
waiting...
waiting...
scanning
SKY907DA : 3,-90,7c:4c:a5:e3:49:89,6
O2wireless892434 : 1,-89,00:1d:68:6c:1e:75,1
dat22wireless : 1,-96,00:09:5b:a9:42:42,9
pibrush : 3,-54,44:94:fc:74:27:a4,11
t is nil
waiting...
waiting...
scanning
pibrush : 3,-54,44:94:fc:74:27:a4,11
O2wireless892434 : 1,-88,00:1d:68:6c:1e:75,1
t is nil
waiting...
waiting...
scanning
pibrush : 3,-50,44:94:fc:74:27:a4,11
O2wireless892434 : 1,-90,00:1d:68:6c:1e:75,1
t is nil
waiting...
waiting...
scanning
SKY907DA : 3,-92,7c:4c:a5:e3:49:89,6
O2wireless892434 : 1,-87,00:1d:68:6c:1e:75,1
pibrush : 3,-51,44:94:fc:74:27:a4,11
t is nil

macfisherman, 2015-01-28:
NodeMCU 0.9.5 build 20150118  powered by Lua 5.1.4
lua: cannot open init.lua
> FILE="bug-with-wdclr.lua" file.remove(FILE) file.open(FILE,"w+") uart.setup(0,9600,8,0,1,0)
> >> > > > 
--Done--
> 

> 
> dofile("bug-with-wdclr.lua")
> waiting...
t is nil
H!�ȩ��F�

�Dǝ�@H��H�
NodeMCU 0.9.5 build 20150118  powered by Lua 5.1.4
lua: cannot open init.lua

macfisherman, 2015-01-28:
here is what airport scan returns on my mac:

```
jeff-mbp:include jeff$ /System/Library/PrivateFrameworks/Apple80211.framework/Versions/A/Resources/airport scan 
                            SSID BSSID             RSSI CHANNEL HT CC SECURITY (auth/unicast/group)
                     xfinitywifi e6:89:xx:xx:xx:xx -84  11,-1   Y  US NONE
                     xfinitywifi e6:3e:xx:xx:xx:xx -88  6       Y  US NONE
     HP-Print-A8-Photosmart 6520 38:ea:xx:xx:xx:xx -81  1       Y  -- WPA2(PSK/AES/AES) 
                             LRJ 00:1d:xx:xx:xx:xx -77  1       Y  US WPA(PSK/TKIP,AES/TKIP) WPA2(PSK/TKIP,AES/TKIP) 
                     xfinitywifi 06:1d:xx:xx:xx:xx -78  1       Y  US NONE
                           5797N f8:e4:xx:xx:xx:xx -81  1       Y  -- WPA2(PSK/AES,TKIP/TKIP) 
                  vachon_network 00:22:xx:xx:xx:xx -83  7,-1    Y  -- WPA(PSK/TKIP,AES/TKIP) WPA2(PSK/TKIP,AES/TKIP) 
                       HOME-44B2 00:1d:xx:xx:xx:xx -85  10      Y  US WPA(PSK/TKIP,AES/TKIP) WPA2(PSK/TKIP,AES/TKIP) 
                       HOME-6855 00:1d:xx:xx:xx:xx -77  11      Y  US WPA(PSK/TKIP,AES/TKIP) WPA2(PSK/TKIP,AES/TKIP) 
                     xfinitywifi 06:1d:xx:xx:xx:xx -73  11      Y  US NONE
                        TO67MDGI 00:26:xx:xx:xx:xx -88  11      Y  US WPA2(PSK/AES/AES) 
                    Brown Family 68:7f:xx:xx:xx:xx -85  11      Y  -- WPA(PSK/AES,TKIP/TKIP) WPA2(PSK/AES,TKIP/TKIP) 
                Guest Network 16 22:c9:xx:xx:xx:xx -82  10      Y  US WPA2(PSK/AES/AES) 
                      Network 16 20:c9:xx:xx:xx:xx -81  10      Y  US WPA2(PSK/AES/AES) 
                    11FX01019366 0c:d5:xx:xx:xx:xx -87  6       N  -- WEP
                              ds 50:46:xx:xx:xx:xx -46  6       Y  -- NONE
                        MacGuest 50:46:xx:xx:xx:xx -46  6       Y  -- NONE
                             mac 50:46:xx:xx:xx:xx -48  6       Y  -- NONE
                            SMac 50:46:xx:xx:xx:xx -48  6       Y  -- WPA2(PSK/AES/AES) 
         DuKarney Family Network 94:10:xx:xx:xx:xx -72  1       Y  -- WPA(PSK/TKIP/TKIP) WPA2(PSK/AES/TKIP) 
                     Bills_2GEXT 6c:b0:xx:xx:xx:xx -81  1       Y  -- WPA2(PSK/AES/AES) 
                            SMac 50:46:xx:xx:xx:xx -45  165     Y  -- WPA2(PSK/AES/AES) 
                 DIRECT-roku-042 b8:xx:xx:xx:xx:xx -56  165     Y  US WPA2(PSK/AES/AES) 

1 IBSS network found:
                            SSID BSSID             RSSI CHANNEL HT CC SECURITY (auth/unicast/group)
                         hpsetup 02:2b:xx:xx:xx:xx -77  6       N  -- NONE
```

asez73, 2015-10-07:
Hi all,
I am juste facing this same issue on version NodeMCU 0.9.5 build 20150318  powered by Lua 5.1.4
...

jmattsson, 2015-11-05:
Closing, see #719.

closed 2015-11-05 by jmattsson

Gennaro001, 2019-08-06:
wifi.sta.getap not work if there aren't AP.
i use:

function listap(t)
    if t==nil then
        print("no net")
        return
    end
    for k,v in pairs(t) do
        print(k.." : "..v)
    end
    print("end scan")
end

wifi.sta.getap(listap)

it work when there are an AP but not when no AP