nodemcu/nodemcu-firmware #124: PWM glitching if tcp server sending or reading data

milikiller, 2015-01-21:
Hi, I try to use my ESP8266 as RC micro telemetry receiver.  I use PWM for servo pulse generator

I use ESP8266 as AP and run TCP server on it.
Communication has made as Ping - Pong.    After client is connected, server send data to client and client send new data back...

Module communicate fast  (one ping - pong is about 15ms)

Problem is PWM, if wifi transfering data, PWM sometimes (cca every 10 seconds) stops and have last value at GPIO, for cca 500ms after this everything back to normal and PWM run fine next cca 10s

Same problem is with timer. If I have timer with 50ms period and transfering data over wifi, timer period is random, sometimes longer than 500ms (10x longer)

I write this code for my simple RC Car with ESP8266, but PWM is problematic and not usable for driving motor.

You can control this code by sending TCP packet at port 2323 to ESP with format "Start;74;512;512;512;Stop"   data is separated by ;  and first is servo, second is PWM for motor1, third is PWM for motor2.

-- PWM frequency for all PWM
PWM_freq = 50

-- Steering servo
pin_servo = 3
servo_pulse_lenght = 74

-- DC motor
pin_hbridge1 = 4
pin_hbridge2 = 2

-- Brake light
pin_brake = 1

-- Mist light
pin_mist_light = 5

-- Main light
pin_main_light = 6

-- Setup servo PWM generator
pwm.setup(pin_servo, PWM_freq, servo_pulse_lenght) 
pwm.start(pin_servo) 

-- Setup DC motor PWM generator
pwm.setup(pin_hbridge1,PWM_freq,0) 
pwm.setup(pin_hbridge2,PWM_freq,0) 
pwm.start(pin_hbridge1)
pwm.start(pin_hbridge2)

-- Setup PWM generator for brake lights
pwm.setup(pin_brake,PWM_freq,0) 
pwm.start(pin_brake)

-- Setup GPIO for mist lights
gpio.mode(pin_mist_light,gpio.OUTPUT)
gpio.write(pin_mist_light,gpio.LOW)

-- Setup GPIO for main lights
gpio.mode(pin_main_light,gpio.OUTPUT)
gpio.write(pin_main_light,gpio.LOW)

print("Creating WIFI AP")

wifi.setmode(wifi.SOFTAP)

cfg={}
cfg.ssid="Auto1"
cfg.pwd="12341234"
wifi.ap.config(cfg)

--cfg = nil

function explode(div,str)
  local pos,arr = 0,{}
  for st,sp in function() return string.find(str,div,pos,true) end do
    table.insert(arr,string.sub(str,pos,st-1)) -- Attach chars left of current divider
    pos = sp + 1 -- Jump past current divider
  end
  table.insert(arr,string.sub(str,pos)) -- Attach chars right of last divider
  return arr
end

function connected(conn)
   print("Client connected")
   
   conn:send("Auto1;123;456")
   connected = 1
   conn:on("receive", function(conn, pl) 
      if string.find(pl, "Start") then
          if string.find(pl, "Stop") then
               local result = explode(";",pl)
                           
               -- Set servo position
               servo_pulse_lenght = tonumber(result[2])/3
               
               pwm.setduty(pin_servo, servo_pulse_lenght)
               pwm.setduty(pin_hbridge1, tonumber(result[3]))
               pwm.setduty(pin_hbridge2, tonumber(result[4]))
                              
               conn:send(servo_pulse_lenght)              
                       
          end    
      end
      
      
   end)
   
   conn:on("disconnection",function(conn) 
      print("disconnect")  
   end)
end

-- Check every 500ms module IP
-- If module have IP start TCP server
tmr.alarm(0, 500, 1, function() 
   if wifi.ap.getip() ~= nil then
   --if wifi.sta.getip() ~= nil then
      print("\r\nWIFI AP Created") 
      print("Module IP")
   
      --print(wifi.sta.getip())
      print(wifi.ap.getip())
      
      sv=net.createServer(net.TCP, 10)
      sv:listen(2323, connected)
      tmr.stop(0)
   else
      uart.write(0,".")
   end
   
end)

vowstar, 2015-01-22:
Current SDK is using software PWM and software timer. When CPU is busy on doing someting, the PWM and timer will stoped until CPU is free. May RTOS SDK can fix this problem, we are developing.

Misha-Mainenko, 2015-11-15:
Cool!

nickandrew, 2016-02-01:
As per @vowstar above, I don't think this is fixable with the current SDK and software PWM. There's no limit to how long the CPU might spend in the SDK, and this will become noticable if your PWM and timers have tight time constraints. Closing.

closed 2016-02-01 by nickandrew