nodemcu/nodemcu-firmware #127: DS18B20P not working

scargill, 2015-01-24:
Such a long time since I did this -  I used to know the difference.. the DS18B20 library works just fine - but does not work at all with the DS18B20P variant.  It SHOULD. Does anyone know what change needs to be made to the library so that both variants will work.   Am I right in saying the 1 second delay normally found is missing? If so could that account for it?

When I've done libraries for this chip in the past, I've always swapped the code over so that the conversion starts AFTER the reading, on the assumption that the temperature will be read regularly and only the first reading will then be missed. That reduces the processing time down to milliseconds - I wonder if anyone has thought of doing this for Lua.

vowstar, 2015-01-25:
I remember in ds18b20's datasheet, a convertion will take 750ms maximum when use 12bit mode. 1 second is safe. 
About DS18B20P variant, could you give me some tip, or sample code?

lmamakos, 2015-01-25:
If you have multiple DS18B20 devices on a single bus, and they are not bus powered, you can also start conversions on all them in parallel, rather than processing them sequentially.  I do this in my application with good success.    I do this to start the conversion:

    ow.reset(pin)
    ow.write(pin, 0xCC, 1)  -- skip ROM
    ow.write(pin, 0x44, 1)  -- convert T
    tmr.delay(1000000)

and then in a loop over all the sensors:

    ow.reset(pin)
    ow.select(pin, addr)
    ow.write(pin, 0xBE, 1)
    data = string.char(ow.read(pin))

scargill, 2015-01-25:
The code is the same. If you do the read and THEN the conversion (which means the read is wrong first time – but then it is wrong first time using the existing code in the Lua library) – as long as you don’t take a reading more than once a second (and no-one surely is going to do that) then you’re ok. I’m taking a reading in a timer every 5 seconds . I’ll likely in practice take a reading every minute.

 

But unless I can suss out what the issue with Lua is I’m thinking of doing a C version.. or rather modifying an existing C version. Not seeing one in the Eclipse setup so I’m off to fine a one-sire library for ESP8266 C.

scargill, 2015-01-25:
Same problem – 1 seconds loop will stop all background communication

NeiroNx, 2015-01-25:
If you use more than one DS18B20 you may need external pull-up resistor.

vowstar, 2015-01-26:
We will add a function named ds18b20.convert(), and use it in tmr.alarm, register an one-short timer call back, every 760ms call ds18b20.read() to get right temperature.

scargill, 2015-01-31:
The code is the same. If you do the read and THEN the conversion (which
means the read is wrong first time – but then it is wrong first time using
the existing code in the Lua library) – as long as you don’t take a reading
more than once a second (and no-one surely is going to do that) then you’re
ok. I’m taking a reading in a timer every 5 seconds . I’ll likely in
practice take a reading every minute.

But unless I can suss out what the issue with Lua is I’m thinking of doing a
C version.. or rather modifying an existing C version. Not seeing one in the
Eclipse setup so I’m off to fine a one-sire library for ESP8266 C.

view it on GitHub.

jmattsson, 2015-11-05:
Closing, see #719. I believe a number of one-wire fixes have gone into dev since this issue was raised.

closed 2015-11-05 by jmattsson