nodemcu/nodemcu-firmware #128: Not reliable gpio timing

dvv, 2015-01-25:
This code
```lua
function meandr()
  local gpio = gpio
  local mode, write = gpio.mode, gpio.write
  local pin = 3
  mode(pin, gpio.OUTPUT)
  while true do
    write(pin, 1)
    write(pin, 0)
  end
  mode(pin, gpio.INPUT)
end
```
produces the following waveform

[image]

while one would expect less variance.

The high pulses are of 8,67us which gives us circa 60Khz at the best conditions. Wonder if the same holds being rewritten in pure C.

vowstar, 2015-01-25:
Hmm...

dvv, 2015-01-25:
NVM, this works ok on another module.
I did this test because the most freq in `pwm` is only 1kHz. Wanted to use `pwm` to blink the diode of wirelessly controlled 38kHz IR remote )

closed 2015-01-25 by dvv

baracudaz, 2015-01-25:
Hi. I am actually interested in doing something similar for my project as well. When I was looking at PWM implementation in NodeMCU it all looked like software implemented PWM doing GPIO bit banging. Happy to be proven wrong. With the max. frequency around 1khz there was not much you not even mentioning desired 38khz. So I am really wondering what results are getting with bit banging in Lua? I would expect Lua loop would be slower the build in software PWM implementation. Could you post your results?

dvv, 2015-01-25:
I squeezed the minimum of 7.8-8.0us pulses. 0.1-0.2us spent in cycle "guts" (`while true do write(1); write(0) end`), so low pulse is longer. Have no oscilloscope to see the fronts.
Since this is a tight loop one must be clever to call `tmr.wdclr()` periodically and this will introduce another errors to meandr. Not mentioning that such loop chokes WiFi. So far, mostly negative feeling... ) (Damn, even minor AVR can do PWM)

baracudaz, 2015-01-25:
I agree 100%. All software PWM implementations are pain. I was also digging into Raspiberry Pi LIRC kernel driver which is also software based. It turns out they are doing some clever time compensation each period to avoid jitter. Anyway my conclusion is to either use a small ARV (anything like Tiny 45/85 would do) all normally carry a hardware PWM or some deep dive C coding would be required to improve PWM. Lua seem to be way too slow for this. What's your thought?

dvv, 2015-01-25:
Well linux is not a RTOS, works with gpios via sysfs (AFAIK), hence can't guarantee timings.
I'd stay with ESP and go to C if I'd time. May turn out that there's no these quirks there since I believe the 1kHz PWM limit is put here just to let the other subsystems breathe.
Going to AVR means at least another communication channel needed, with all its negatives.