nodemcu/nodemcu-firmware #139: Non-blocking UART transmit

dvv, 2015-01-27:
https://github.com/nodemcu/nodemcu-firmware/blob/master/app/driver/uart.c#L79-L85 essentially polls for TX buffer empty in manner they do with AVR:
```c
// char gone on wire
OS_INTERRUPT void USART_UDRE_vect(void)
{
  // anything else in output buffer?
  if (txhead != txtail) {
    uint8_t tmptail = (txtail + 1) & UART_TX_BUFFER_MASK;
    txtail = tmptail;
    // disable reader
    UCSR0B &= ~_BV(RXEN0);
    // load next char from output buffer
    UDR0 = uart_tx_buf[tmptail];
  // output buffer empty
  } else {
    // disable TX ready interrupt
    UCSR0B &= ~_BV(UDRIE0);
  }
}

OS_INTERRUPT void USART_TX_vect(void)
{
  // char transmitted, enable reader
  UCSR0B |= _BV(RXEN0);
}

static void uart_putc(char c)
{
  // put to output buffer
  uint8_t tmphead = (txhead + 1) & UART_TX_BUFFER_MASK;
  while (tmphead == txtail) {}
  uart_tx_buf[tmphead] = c;
  txhead = tmphead;
  // enable TX ready interrupt
  UCSR0B |= _BV(UDRIE0);
}
```

Otherwise printing long lines (e.g. TCP payload) hangs the CPU.

vowstar, 2015-02-16:
We are doing RTOS version NodeMCU, it may solve this issue.

jmattsson, 2015-11-05:
Closing, see #719.

closed 2015-11-05 by jmattsson