nodemcu/nodemcu-firmware #140: string.format not working since 0127(float version)

nodemcu, 2015-01-27:
from esp8266.com
```lua
t=1
= string.format("%i, ",t)
```

nodemcu, 2015-01-27:
update, fixed.
pls test.

unwin, 2015-01-31:
I think the behavior of string.format's handling of '-' has changed.  string.format("%-10s",k) produces 10s as output. Removing the '-' produces the contents of the string formatted to 10 spaces, right justified.

dvv, 2015-01-31:
optimized printf in 0.9.5?

vowstar, 2015-01-31:
Because SDK's printf not support floating number, we write one but not beautiful yet.

closed 2015-02-11 by vowstar

vowstar, 2015-02-11:
[image]

GeoNomad, 2015-02-11:
Can you please try string.format("%-10s",'this')

That is the part that has been breaking LuaLoader's file.list and Survey buttons.

It has been printing 

-10s

Thanks.

vowstar, 2015-02-12:
[image]
It out put this. Is it right?

nodemcu, 2015-02-12:
I got this, I think it is right! well done
```lua
> i=-1
> =string.format("%-10f,end", i)
-1.000000 ,end
> =string.format("%-10d,end", i)
-1        ,end
>
```

GeoNomad, 2015-02-13:
Yes, it is fixed. Thanks guys!

commit f3540ffe (2015-02-11): Fixed many floating point bug.
Fixed #148 String:format Floating point, width.precision not working.
Fixed #140 string.format not working since 0127(float version) #140.