nodemcu/nodemcu-firmware #141: Build nodemcu_20150127.bin issues "not enough memory"

CHERTS, 2015-01-28:
I uploaded files lua_modules\bmp085\bmp085.lua and init.lua
Init.lua file contains:
```lua
bmp085 = require("bmp085")
sda = 4
scl = 3
bmp085.init(sda, scl)
p = bmp085.getUT(false)
print(p)
p = bmp085.getUT(true)
print(p)
bmp085 = nil
package.loaded["bmp085"]=nil
```
After restarting the ESP module the message:
NodeMCU 0.9.5 build 20150127 powered by Lua 5.1.4
lua: error loading module 'bmp085' from file 'bmp085.lua':
not enough memory

I check the availability of files, they are
filelist = file.list () for filename, filesize in pairs (filelist) do print (filesize .. "" .. filename) end
285 init.lua
5194 bmp085.lua

I also check the contents of the file and init.lua bmp085.lua, with them everything is fine:
file.open("init.lua") print(file.seek("end", 0)) file.seek("set", 0) repeat local line=file.readline() if line then line=(string.gsub(line,"\n","")) print(line) end until not line file.close()
285
bmp085 = require("bmp085")
sda = 4
scl = 3
bmp085.init(sda, scl)
p = bmp085.getUT(false)
print(p)
p = bmp085.getUT(true)
print(p)
bmp085 = nil
package.loaded["bmp085"]=nil

I think this is a bug in the firmware.

joaodh92, 2015-01-29:
i make one code to control lcd pdc8455, but when i insert a ASCI table for caracters, i got the message "not enough memory". somebody  can help me?

vowstar, 2015-02-16:
Please try to compile your LUA code in new version, it may help you save memory.

dvv, 2015-02-16:
Same with https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_examples/yet-another-bmp085.lua -- can the flaw be inside `i2c`?

kireevco, 2015-02-22:
Same thing started happening with DHT11 inside listen event
srv:listen(80,function(conn) end

I'm loading a module, and this is happening.
It doesn't do that when loaded standalone from Lua prompt.
There is plenty of memory available when my server app is loaded as well.

Any version that doesn't do that?

dariocdj, 2015-03-14:
The same thing also with bmp085 or bmp180. If you use it alone from bmp085.lua it works. When you try to load lib from init.lua it gives out of memory error....

Is it depends from i2c or what? If i dofile bmp085 to get temp or press it works, if i start tcp server in init.lua it crash in lua: error loading module 'bmp085' from file 'bmp085.lua':
not enough memory

tried also to create a simple init.lua with 1 line only calling bmp085 and give me not enough memory.
Tried also to format esp.
Tried with last firmware and with 0.9.4.... nothing

Help ! tnx

rtenklooster, 2015-05-21:
Anyone found a solution for this problem? i have tried the latest firmware version, but still not enough memory.. would be very happy to get this sensor  working on my esp-201

dariocdj, 2015-05-21:
I've solved with nodemcu_20150123

TerryE, 2015-08-21:
I will assume that since we've had a posted solution and this issue is inactive that we should close it.

closed 2015-08-21 by TerryE