nodemcu/nodemcu-firmware #144: Expose json

dvv, 2015-01-28:
Please, consider exposing `str = json.stringify(tbl)` and `tbl = json.parse(str)`, or alike.
TIA

vowstar, 2015-01-29:
Yes, with json, eveything comes easy.

marcoskirsch, 2015-02-02:
There's plenty of example code floating around to go from lua table to JSON and back. Why not use that?

dvv, 2015-02-02:
json stuff already in sdk, needs mostly only proper exposing

iMarvinS, 2015-02-12:
Is someone here able to expose the json API?

alex-march, 2015-02-23:
Agree, would be a neat way to handle json in nodemcu, otherwise there are some lightweight options in pure lua.

zhivko, 2015-03-12:
Would be nice to have json exposed as module.
Which existing module would be most apropriate candidate for rework, so we could have json module?

nodemcu, 2015-03-16:
cjson ported to nodemcu, currently in json branch. please fork and test.
cjson manual is [here](https://github.com/mpx/lua-cjson).
and also download latest build for cjson to test [download](https://github.com/nodemcu/nodemcu-firmware/releases/latest), maybe need some time for travis-ci to build?

dvv, 2015-03-16:
great!

alex-march, 2015-03-16:
Good stuff, thank you!

marcoskirsch, 2015-03-16:
It's great to have this. But some applications don't need JSON and the firmware as is makes for an extremely memory constrained environment. Please make it so it's easy to compile in/out based on a single #define. That request holds for most non essential functionality... thanks!

dvv, 2015-03-16:
We should definitely stab the knife to exploring loadable c modules support.
This may be of concern even for intrinic bit/table modules.

nodemcu, 2015-03-17:
@marcoskirsch 
[/app/include/user_modules.h](https://github.com/nodemcu/nodemcu-firmware/blob/json/app/include/user_modules.h)
one line #define for each module.

closed 2015-03-18 by nodemcu