nodemcu/nodemcu-firmware #145: node.readvdd33() causes reboot while wifi is connected

Mehrius, 2015-01-28:
Hello everybody,

My module reboots if I run node.readvdd33() while wifi is connected.
It runs fine when wifi is not connected.

Do any of you also have this problem?

vowstar, 2015-01-29:
In sta mode when connect wifi it will reboot , in ap or stationap or not connect wifi it will have no problem. It is SDK 0.9.5's bug.

Mehrius, 2015-01-29:
Thank you,

Not being nodeMCU related I guess I should close this issue.

closed 2015-01-29 by Mehrius

vowstar, 2015-02-16:
The 0.9.6_b1 SDK import a new API system_get_vdd33(), but it always return 65535.
May be espressif want fix it, we are waiting 0.9.6 finish beta.

vowstar, 2015-05-19:
https://github.com/nodemcu/nodemcu-firmware/commit/fc8fa2cc915aea3687b068eb0355bb84b2fc113d