nodemcu/nodemcu-firmware #148: String:format Floating point,  width.precision not working

Rotma, 2015-01-29:
> t=2.345
> = string.format("%02f",t)
2.345
> = string.format("%02.2f",t)
2f
> = string.format("%02d",t)
02
>

sej7278, 2015-02-03:
isn't this fixed since https://github.com/nodemcu/nodemcu-firmware/commit/1346cbbc38976ea49076755969710bfea75dbf55 ?

Rotma, 2015-02-04:
Example was with build 20150126
Latest bin 20150127 string.format("%02f",t) returns nothing.

vowstar, 2015-02-11:
[image]

closed 2015-02-11 by vowstar

willd, 2015-02-13:
Works like a charm, thanks!

Rotma, 2015-02-14:
Still one small problem, %03.2f should produce 002.04... (preceding zero's) like %03d 
For the rest it's working ok.

NodeMCU 0.9.5 build 20150213  powered by Lua 5.1.4
> a=2.045
> print(string.format("%03.2f\n",a))
2.04
> print(string.format("%03d",a))
002

TheoHarbers, 2015-02-14:
Not entirely true it should be:
> a=2.045
> print(string.format("%03.2f\n",a))
> 2.05

vowstar, 2015-02-15:
Hi Rotma, it is not bug. please test it use C 's printf and read c99's field width definition of printf in floating case.
If you want 0.002.04, please use print(string.format("%06.2f\n",a))
[image]

Rotma, 2015-02-15:
Vowstar, you are right my mistake.
But another thing i noticed  after the comment of Theo is that roundig up starts at 6.
This should be normally at 5 i tought.

> a=2.045
> b=2.046
> print(string.format("%05.2f - %05.2f",a,b))

02.04 - 02.05

vowstar, 2015-02-16:
I think it is not an error.
"For the GNU C library, the rounding rule used by printf() is "bankers rounding" or "round to even". This is more correct than some other C libraries, as the C99 specification says that conversion to decimal should use the currently selected IEEE rounding mode (default bankers rounding)."
See also:
http://stackoverflow.com/questions/10357192/printf-rounding-behavior-for-doubles
http://linuxgazette.net/144/misc/lg/a_question_of_rounding_in_issue_143.html
http://www.gnu.org/software/libc/manual/html_node/Rounding.html
And  "bankers rounding" or "round to even" in Wikipedia:
http://en.wikipedia.org/wiki/Rounding
Please read it carefully.

The result on build 20150213 NodeMCU:
[image]

I have tested this behavior on my debian sid, it have same results:
[image]

I have tested this behavior on my centos, it have same results:
[image]

```
> a=2.045
> print(string.format("%05.2f",a))
02.04
> a=2.055
> print(string.format("%05.2f",a))
02.06
>
```

commit f3540ffe (2015-02-11): Fixed many floating point bug.
Fixed #148 String:format Floating point, width.precision not working.
Fixed #140 string.format not working since 0127(float version) #140.