nodemcu/nodemcu-firmware #150: Feature Request: mDNS

dereulenspiegel, 2015-01-30:
Hello,

nodeMCU is a great firmware to quickly realize small IoT projects. To make this even easier it would be nice to be able to respond to mDNS requests, so we can discover nodeMCU devices in the network.

ersinpw, 2015-01-31:
I totally agree

sej7278, 2015-02-03:
i'd certainly want an option to disable this if someone does implement it, yuck multicast!

dereulenspiegel, 2015-02-03:
I would have expected that this is a feature which can enabled and disabled via the user_config.h or has to be explicitly called via a LUA API call and if you don't call this API the node doesn't react to multicast packages.
It would also be nice to be able to announce services and not only an A record, so protocol and port can be discovered.

mkarliner, 2015-02-04:
+1 from me, although I'm still learning about zeroconf
Looking at the last couple of days of issues, with lots
of conversations about features, a good config system
would seem to be a good idea, so we can all build our
own 'perfect' nodemcu

sej7278, 2015-02-04:
@mkarliner you bring up a good point, all of these macro's should really be compile-time flags in the Makefile, so we don't have to edit a bunch of header files

marcelstoer, 2015-06-08:
> a good config system would seem to be a good idea, so we can all build our own 'perfect' nodemcu

@mkarliner I developed http://frightanic.com/nodemcu-custom-build/ for that. It might be useful for your needs, too.

aguaviva, 2015-07-27:
any updates on this one?

saeugetier, 2016-01-27:
Issue is still open. Is there any progress implementing mDNS for nodemcu?

DesperateProgrammer, 2016-01-27:
I am working on a user module for mDNS/Bonjour/ZeroConf welk at least in my sparse sparetime.

Progress can ne found at http://desperate-programmers.com/2016/01/24/name-resolving-demo-mdns-reponder/

When the LUA Interface is set i'll try to get it into this git (dont know git as much as svn yet.

saeugetier, 2016-01-27:
Thanks. I ported your bonjour.c into my fork of nodemcu. It is working now...

pjsg, 2016-01-28:
I have a mdns lua module that acts as a server and can advertise the name of the board and a single service. Needs a bit more testing, but appears to work.

I'll submit a PR in the next couple of days.   

The documentation is at https://github.com/pjsg/nodemcu-firmware/blob/mdns/docs/en/modules/mdns.md -- let me know if it does what you need. This is a pretty thin layer over the Espressif provided mdns implementation, so there is not a lot of room for enhancement without significant work. However it does provide the basic capabilities.

pjsg, 2016-01-29:
I've submitted PR #988 to provide this implementation.

ersinpw, 2016-01-29:
@pjsg thanks!

pjsg, 2016-03-05:
This has now been merged and is now in the dev branch.

closed 2016-03-05 by pjsg