nodemcu/nodemcu-firmware #152: Multiple MQTT Subscriptions

Helix99, 2015-01-30:
If I parse multiple subscribe commands I either get a sending in process panic :

    m:connect("192.168.11.10", 1883, 0, function(conn)
         print("connected")
         m:subscribe("test",0, function(conn) print("subscribe test")
         end)
         m:subscribe("test2",0, function(conn) print("subscribe test2")
         end)
    end)

response :
> dofile('mqtt_test.lua')
> connected
PANIC: unprotected error in call to Lua API (mqtt_test.lua:21: sending in process)

of if I nest the subscriptions, they are repeated :

    m:connect("192.168.11.10", 1883, 0, function(conn)
         print("connected")
         m:subscribe("test",0, function(conn) print("subscribe test")
              m:subscribe("test2",0, function(conn) print("subscribe test2")
              end)
         end)
    end)

Response :
> dofile('mqtt_test.lua')
> connected
subscribe test
subscribe test2
subscribe test2

I can get round the issue with the following code to allow the subscriptions to process :+1: 
--connect to the broker
m:connect("192.168.11.10", 1883, 0, function(conn)
     print("connected")
     mqtt_sub() --run the subscription function
end)

function mqtt_sub()

     -- if we have subscribed to all topics in the array, run the main prog
     if table.getn(topics) < current_topic then
     
          run_main_prog()
         
     else
          --subscribe to the topic
          m:subscribe(topics[current_topic],0, function(conn)
               print("subscribe")
          end)
          -- increment the variable of the current topic for next loop     
          current_topic = current_topic + 1
          --set the timer to rerun the loop
          tmr.alarm(0, alarm_delay, 0, function()
               mqtt_sub()
          end)
     end
end

--main program to run after the subscriptions are done
function run_main_prog()
     print("Hello World")
end

Would it be possible to add some form of queue for the MQTT subscriptions??

Suxsem, 2015-01-30:
Or maybe make mqtt operations to stop the code until they are done...
This bug actually makes mqtt module useless...

closed 2015-02-03 by vowstar

crapthings, 2015-03-25:
Can we fix publish too ?

https://github.com/nodemcu/nodemcu-firmware/issues/146

AndyKirby, 2015-03-25:
+1 for semaphore, or at least an api call that lets you test whether
sending is currently in progress or not before attempting another.

Yes it is always possible to work around this in script, but better
being part of the API, and documented somehow.

With the API extension to return a boolean on busy, you can do a
semaphore or a queue according to your needs/app without having to
implement that bit in every bit of code you write. It is a bit of a
basic essential.

BTW I think the MQTT snippet on the NodeMCU page either needs fixing or
removing. It doesn’t work as writ and has lead a number of newbies I
assist to the incorrect conclusion that NODEMCU and MQTT is
irretrievably broken and not getting fixed.

IMHO

Kirbs

mkarliner, 2015-03-25:
You could just submit a pull request for the README. 
I agree that the snippet doesn't work, but I think it should,
so the conclusion is a reasonable one.