nodemcu/nodemcu-firmware #162: readVDD33() crashes chip

Manawyrm, 2015-02-01:
Hello,

the new function node.readvdd33() sometimes crashes and reboots the chip. 
https://github.com/nodemcu/nodemcu-firmware/commit/4af087312211a5d699c86c3e64fb7ba99200ecbd

https://screenshot.tbspace.de/lzavpeyiqgd.png

Any ideas? 
Best regards,
Tobias

vowstar, 2015-02-02:
In sta mode when connect wifi it will reboot , in ap or stationap or not connect wifi it will have no problem. It is SDK 0.9.5's bug. We are waiting for Espressif's new SDK patch.

See also
https://github.com/nodemcu/nodemcu-firmware/issues/145
http://www.esp8266.com/viewtopic.php?f=18&t=1417

vowstar, 2015-02-16:
The 0.9.6_b1 SDK import a new API system_get_vdd33(), but it always return 65535.
May be espressif want fix it, we are waiting 0.9.6 finish beta.

diaconom, 2015-03-23:
Any word on when this feature will be fixed ?

vowstar, 2015-03-25:
Yes, I have a dirty idea. I think we can add protection before use readvdd33(), before using it we will disconnect Wi-Fi, and after use it we will reconnect it.

chadouming, 2015-03-25:
This could be an idea, however, if you go that way could you make it something like readvdd33_safe(). Or you could check if device is in station before disconnect/reconnect. Last suggestion, instead of disconnecting/reconnecting, wouldn't switch from station to stationap be better ?

nodemcu, 2015-03-25:
I was considering remove this api, 
any must use case?

chadouming, 2015-03-25:
battery powered project ? 

temp.bbqdroid.org

nodemcu, 2015-03-25:
can a battery power apply directly to esp8266 vcc pin?

chadouming, 2015-03-25:
i'm using LiFePo4 battery, which is pretty much perfect for this.

GeoNomad, 2015-03-25:
*Yes, I have a dirty idea. I think we can add protection before use readvdd33(), before using it we will disconnect Wi-Fi, and after use it we will reconnect it.*

Seems that would lead to more bug reports rather than fewer.

I suggest leaving the API but documenting when it can't be used. Or testing wifi status and returning nil when measuring is not possible.

Battery power is certainly used and being able to shut down when LiPo batteries reach a specified voltage is an important use case for protecting the battery.

joefly888, 2015-05-27:
i noticed if i run this program it works fine
 http://captain-slow.dk/2015/04/16/posting-to-thingspeak-with-esp8266-and-nodemcu/

but if I issue a manual command in esplorer with =node.readvdd33() it will mostly restart but occasionally work

commit 3dd1ac37 (2015-03-26): Move node.readvdd33 to adc.readvdd33, fix #162.
closed 2015-06-17 by vowstar