nodemcu/nodemcu-firmware #163: PANIC: unprotected error in call to Lua API (attempt to call a nil value)

michal2229, 2015-02-01:
Hello,
I am doing a project, in which one can control TV or other IR receiver from web page hosted on ESP. 
ESP parses data from GET request sent from web page. Then ESP sends message to atmega and finally atmega sends IR code trough IR led. Everything works fine for about 85% of requests. For these 15% of cases ESP says: 
'PANIC: unprotected error in call to Lua API (attempt to call a nil value)' 
and restarts. 
It happens more frequent when I access the page from my smartphone.
Is there a way to avoid it?

Here is my ESP code:
```lua
wifi.setmode(wifi.STATIONAP)
cfg={}; 
cfg.ssid="ESP8266_" .. wifi.ap.getmac(); cfg.pwd="mypassword"; 
wifi.ap.config(cfg); 

server = net.createServer(net.TCP) 
server:listen(80, function(conn)
	conn:on("receive", function(conn, receivedData) 
		if receivedData then
			if string.find(receivedData, "GET /ir_code") then
				message_to_uc = string.sub(receivedData, string.find(receivedData, "GET /ir_code.- ")); 
				-- message parsed and printed to UART (Atmega receives it, parses and sends trough IR)
				if message_to_uc then print(message_to_uc .. ' ' .. node.heap()) end -- node.heap() added to debug
			end
			if (string.find(receivedData, "HTTP") ~= nil) then -- sending data to browser
				html = "<html><head><title>myProject</title><head><body>"
					.. "<form action='ir_code' method='get'> "
					.. "<input id='type' name='codeType' value='SAMSUNG'> "
					.. "<input id='value' name='codeValue' value='0xE0E040BF'> "
					.. "<button id='send'>send</button> "
					.. "</form> "
					.. "timer: " .. tmr.now()/1000 .. "s <br>"
					.. "heap: " .. node.heap() .. "B"
					.. "</body></html>"
				conn:send("HTTP/1.1 200 OK\r\n")
				conn:send("Content-Length:" .. string.len(html) .. "\r\n")
				conn:send("Connection:close\r\n\r\n")
				conn:send(html);
			else       
				conn:send(tmr.now()/1000 .. " unknown data "); 
			end
		end
	end) 
	conn:on("sent", function(conn) 
		collectgarbage() 
	end)
end)
```

And this is the output from ESP:

NodeMCU 0.9.5 build 20150127  powered by Lua 5.1.4
GET /ir_code?codeType=SAMSUNG&codeValue=0xE0E040BF  12968
GET /ir_code?codeType=SAMSUNG&codeValue=0xE0E040BF  12752
GET /ir_code?codeType=SAMSUNG&codeValue=0xE0E040BF  12752
GET /ir_code?codeType=SAMSUNG&codeValue=0xE0E040BF  12752
GET /ir_code?codeType=SAMSUNG&codeValue=0xE0E040BF  12752
GET /ir_code?codeType=SAMSUNG&codeValue=0xE0E040BF  12752
GET /ir_code?codeType=SAMSUNG&codeValue=0xE0E040BF  12288
GET /ir_code?codeType=SAMSUNG&codeValue=0xE0E040BF  12264
GET /ir_code?codeType=SAMSUNG&codeValue=0xE0E040BF  12512
GET /ir_code?codeType=SAMSUNG&codeValue=0xE0E040BF  11722
GET /ir_code?codeType=SAMSUNG&codeValue=0xE0E040BF  12436
GET /ir_code?codeType=SAMSUNG&codeValue=0xE0E040BF  12144
GET /ir_code?codeType=SAMSUNG&codeValue=0xE0E040BF  11768
PANIC: unprotected error in call to Lua API (attempt to call a nil value)

NodeMCU 0.9.5 build 20150127  powered by Lua 5.1.4

Thanks in advance.

MarsTechHAN, 2015-02-05:
The feedback "PANIC: unprotected error in call to Lua API (attempt to call a nil value)"
means you use a nil value(without define)check your program where use a value gathered from the GET data

mkarliner, 2015-02-05:
You may want to read the sections on exception handling in the Lua manual.
http://www.lua.org/pil/8.4.html

For instance, to catch exceptions sending a value to MQTT, I do...

     if pcall(readtemp) then
          print("Temp sent OK")
     else
          print("Temp err" )
     end

function readtemp() 
      m:publish("temperature",temp,0,0, function(conn) print("sent") end)
 end

michal2229, 2015-02-06:
Thank You for Your responses. However I could not find any 'nil value' variables and I had no idea how to apply pcall() to my code in any reasonable way :) 
Anyway, changing the line:
```lua
server = net.createServer(net.TCP)
```
to:
```lua
server = net.createServer(net.TCP, 1) 
```
fixed the problem :)

MarsTechHAN, 2015-02-07:
...Next time...You need to check the API Document carefully...
TCP Server need to set a timeout seconds.
(We also recommend the author to check that value before using...)

nodemcu, 2015-02-07:
My fault, API is not clear.
net.createServer has a default time out for connections.
as a http server, better close connection when page is sent.

michal2229, 2015-02-07:
Well... I overlooked it... 
Thank You all for help, it is a bit clearer to me now :)

closed 2015-02-11 by nodemcu

vishwaraj00, 2016-02-17:
If you do not use the HTTP module in the firmware then this error comes.

kilhankulhe, 2016-06-29:
How can I Post something with JSON format, and with X-Auth-Token as Header?

TerryE, 2016-07-03:
@kilhankulhe, this is a Q that has nothing to do with this closed issue and best and on an application support forum.

Craig1516, 2016-09-04:
vishwaraj00, Thanks for the HTTP module firmware comment!  That solved my stability problem.  I was getting resets and errors several times a minute.  Your solution solved it.  For others: (Build your custom firmware at https://nodemcu-build.com/  @vishwaraj00