nodemcu/nodemcu-firmware #169: gpio.mode(gpio.INPUT or gpio.PULLUP) not going back to gpio.PULLUP mode after pulldown

clixx-io, 2015-02-02:
I have a button that's floating, and when press, passes through to Ground.

So I use gpio.mode(4,gpio.INPUT or gpio.PULLUP) and then =gpio.read(4) which gives "1". Good

Then I press the button, and execute gpio.read(4) and the result is "0". Good.

However, the gpio.read(4) continues to output "0" even when the button is released.

If I execute another gpio.mode(4,gpio.INPUT or gpio.PULLUP) subsequent reads never return to "1". 

The only solution that I have found is to reboot the module to read the button again.

dvv, 2015-02-02:
`gpio.mode(4,gpio.INPUT or gpio.PULLUP)` means `gpio.mode(4,true)` which effectively means `gpio.mode(4,gpio.OUTPUT)`.
To set it a pullup input use `gpio.mode(4,gpio.INPUT, gpio.PULLUP)` -- notice _the third_ parameter control pulls.

clixx-io, 2015-02-02:
Oh look at that - Perfect !

Thanks.

You can close this now - much appreciated.

closed 2015-02-04 by nodemcu