nodemcu/nodemcu-firmware #173: missing wifi connected event

Suxsem, 2015-02-03:
Hi. Maybe i'm missing something... but how can i write code that is executed right after wifi connect?

I would love something like
`wifi.sta.config("SSID","password", function() print("connected" end)

sej7278, 2015-02-03:
you have to check for your ip address being not nil in a loop

```lua
tmr.alarm(0, 1000, 1, function()
    if wifi.sta.getip() ~= nil then
        tmr.stop(0)
    end
end)
```

Suxsem, 2015-02-03:
not the best solution but working, thanks :)

MarsTechHAN, 2015-02-04:
The reason why some of the function had call back because the ESP SDK has an option to deliever a function point.
Also your program going on, process must be done one by one but mutipuly thread(I mean...Like)
You can use some your  own way.

Suxsem, 2015-02-04:
okok, got it :)
another question... is possible to get the current wifi strength in station mode?
i know i can do wifi.sta.getap(), but i would like to get the strength of the CURRENT connected ap :)

dvv, 2015-02-04:
I would also like to _set_ the strength. Don't like those super mW all around the room, eggs are of priority )

TerryE, 2015-12-12:
Implemented

closed 2015-12-12 by TerryE