nodemcu/nodemcu-firmware #174: Floating point behaviour

TheoHarbers, 2015-02-03:
Trying to produce numbers with just one decimal i found that does not work for some of them.
Example:
   print( 101/10, 102/10, 103/10, 104/10, 105/10 )
   print( 106/10, 107/10, 108/10, 109/10 )
Output:
  10.0999999999 10.199999999 10.3 10.4 10.5
  10.599999999	10.699999999	10.8	10.9

FourQuarks, 2015-02-04:
The best way to tackle this behaviour is to disable the floats.
The activation of the floats are reducing the amount of available SRAM
and there are numerous side effects (e. g. string.format). The benefit of floats is
very small because it is not possible to  enable the LUA maths routines due
to lack of memory.

The ESP8266 is simple to small to handle floating point arithmetics.

dvv, 2015-02-04:
`10.0(9)` is indeed trivially equal to `10.1` because `10.0(9) * 10 - 10.0(9) = 100.9(9) - 10.0(9) = 90.9 = 10.1 * 9` )
But +1 to remove floats by default.

TheoHarbers, 2015-02-04:
Floats are very useful when it is working correctly, otherwise having more free memory could be of more use. 

While floats are still enabled this workaround is working for me:

local temp = 101
print( "T=" .. (temp-temp%10)/10 .. '.' .. temp%10)

Output: 10.1

dvv, 2015-02-04:
But they _are_ working properly. Wonder what displays big lua or nodejs for the same samples?
What if temp = -101?

nodemcu, 2015-02-04:
vote here:)
remove float support in pre_build or not.
but some lua module will not compatible with int32 and float.

Manawyrm, 2015-02-04:
I think it should be disabled by default (and my commit regarding ds18b20 should be deleted..).
It's not really needed and takes up valuable SRAM.

dvv, 2015-02-04:
IMO: i) floats should float away. ii) ds18b20 et al. should _not do any_ formatting. Just return integer centigrads and let the caller convert it or not.
The same applied to other modules -- they contain formatting, they contain hardcoded error printing and behave unfriendly (what if i use stdout for communication?)
The standard proven practice is to make functions to either `return result`, or `return nil, error`

TheoHarbers, 2015-02-04:
Often sensors will upload their data to servers not owned by the same person and require values in human readable format with or without decimals. There is several ways to do this with or without floats. I would vote for no floats for the moment.

dvv, 2015-02-04:
True. But I believe one should not write a library module that does both measurement and upload in one atomic operation.
Instead of using cryptic
```lua
require("foo").measureAndUpload()
```
use
```lua
local foo = require("foo")
local result, error = foo.measure()
if result then
  upload(("%..."):format(result))
else
  print("-ERR", error) -- or may be I as user prefer print("-ОШИБКА:" .. error), whatever 
end
```
This way allows to divide and conquer easily because of clear separation into a pure function and action.

vowstar, 2015-02-05:
http://www.esp8266.com/viewtopic.php?f=18&t=1436

vowstar, 2015-02-11:
Fixed.
https://github.com/nodemcu/nodemcu-firmware/commit/f3540ffe29877ecfb37733a56625671bcd916122

[image]

closed 2015-02-11 by vowstar

vowstar, 2015-02-11:
[image]