nodemcu/nodemcu-firmware #175: Hardcoded HSPI in SPI.C

mtornblad, 2015-02-04:
In /app/driver/spi.c it looks lite HSPI is hardcoded in one place where there probably shoud be a parameter from the function?

WRITE_PERI_REG(SPI_W0(HSPI), *data);

from...

void spi_mast_byte_write(uint8 spi_no, uint8 *data)
{
    if(spi_no>1) 		return; //handle invalid input number
    while(READ_PERI_REG(SPI_CMD(spi_no))&SPI_USR);
    WRITE_PERI_REG(SPI_W0(HSPI), *data);
    SET_PERI_REG_MASK(SPI_CMD(spi_no), SPI_USR);
    while(READ_PERI_REG(SPI_CMD(spi_no))&SPI_USR);
    *data = (uint8)(READ_PERI_REG(SPI_W0(spi_no))&0xff);
}

nodemcu, 2015-02-04:
don't get it.
what do you mean by "hardcoded in one place"

mtornblad, 2015-02-04:
This line..

https://github.com/nodemcu/nodemcu-firmware/blob/master/app/driver/spi.c#L137

it contains the defined value of HSPI instead of the variable spi_no variable
like all the other lines does!

//Mattias

closed 2015-02-04 by nodemcu

mtornblad, 2015-02-04:
> Closed #175 <https://github.com/nodemcu/nodemcu-firmware/issues/175> via
> 1740841

Is trying to understand how SPI works and using your code as a starting
point...

Found this line also..

https://github.com/nodemcu/nodemcu-firmware/blob/master/app/driver/spi.c#L106

//Mattias

commit 1740841e (2015-02-04): fix #175