nodemcu/nodemcu-firmware #176: Please push PR to branch "dev" instead of master.

nodemcu, 2015-02-04:
Thank you everyone for contributing to this project.
I have made branch "dev" up-to-date with master branch.
new features or some major fixes can first push to this "dev" branch.
note that pre_build bin in this "dev" branch is never up-to-date.
for ones want to test new features, please pull "dev" branch and compile.

Suxsem, 2015-02-04:
thanks! :) please don't forget to build new pre_build! :)

vowstar, 2015-02-05:
Ok, I agree with it =)

funshine, 2015-11-21:
I would like to open this issue again.
there's many PRs in master ahead of dev branch. we are now going to sort things out.

nodemcu, 2015-11-21:
I would like to open this issue again.
there's many PRs in master ahead of dev branch. we are now going to sort things out.

karrots, 2015-11-21:
#575 & #675 were cherry picked over to 'dev'.

nodemcu, 2015-11-21:
great, once all done, we will add this to guideline and can be closed.

TerryE, 2015-11-21:
I'll sort out any residual manual merges tomorrow.

marcelstoer, 2015-11-21:
@karrots Thanks for checking.
@nodemcu It already _is_ in the guidelines, see https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md.

TerryE, 2015-12-12:
We now have this as a contribution guideline which is being actively enforced by the committers so we can close this as an issue.

PR #256 (2015-03-01): Move contributing guidelines to repository
Moves guidelines for contribution from issue #176 to a file in the repository. Github will show a message with this file to contributors, increasing its visibility.

Closes #176
commit 2b5ee453 (2015-03-01): Move contributing guidelines to repository

Moves guidelines for contribution from issue #176 to a file in the repository. Github will show a message with this file to contributors, increasing its visibility.

Closes #176
closed 2015-12-12 by TerryE