nodemcu/nodemcu-firmware #177: Improving GPIO toggling speed

raz123, 2015-02-05:
I have ran a few test to see how fast one could toggle the GPIO line in LUA and here are my results:

nodemcu (CPU @ 80 Mhz): ~10 Khz
nodemcu (CPU @ 160 Mhz): ~20 Khz
nodelua (CPU @ 80 Mhz): ~44 Khz

Is there a good reason for which nodemcu is 4x slower than its counterpart?
Do you think it would be possible to improve the performance of the GPIO handling in nodemcu?

dvv, 2015-02-05:
code?

raz123, 2015-02-05:
The code to toggle the GPIO line was simply:
```lua
gpio.mode(2, gpio.OUTPUT)

for i = 1,10 do

gpio.write(2, 1)
gpio.write(2, 0)

end
```

nodemcu's gpio code to write to a pin is the following:

```lua

```

dvv, 2015-02-05:
I squeezed circa 60kHz of it.
The clue is the same that I keep telling here: always do local-ize global variables:
```lua
do
  local mode, write, OUTPUT = gpio.mode, gpio.write, gpio.OUTPUT
  mode(2, OUTPUT)
  for i = 1,10 do
    write(2, 1)
    write(2, 0)
  end
end
```

Mind to retry and post the results of this code? TIA

raz123, 2015-02-05:
dvv, woah, that is quite interesting! Here are my results, using your code:

~51 kHz on the nodemcu 20150127 firmware (gpio pin = 4)
~52 kHz on the nodelua firmware (gpio pin = 4)
~63 kHz on the jrahlf 20150127 fork (gpio pin = 2)

Thanks for the tip!

dvv, 2015-02-05:
yw

raz123, 2015-02-05:
Just landed on this document that discusses optimizing Lua code. Will leave it here for those who happen to stumble on this thread: http://www.lua.org/gems/sample.pdf

dvv, 2015-02-05:
Yup. Though this one imo should be linked to the main project's page ) Lua's very simple yet has some known surprises everyone should know.

urmilparikh, 2015-08-17:
Very helpful tip, @dvv, Thanks!

TerryE, 2015-08-17:
It's in my [unofficial FAQ](http://www.esp8266.com/wiki/doku.php?id=nodemcu-unofficial-faq), but note that these are nearly all optimisation for time rather than space.  You typically need the latter on the esp8266.

closed 2015-08-17 by TerryE