nodemcu/nodemcu-firmware #179: gpio.trig() won't work

ystrem, 2015-02-05:
Hi, I tried this example, but when i push button it count to 4 000 000 and then reboot.

```
use pin 0 as the input pulse width counter
    pulse1 = 0
    du = 0
    gpio.mode(1,gpio.INT)
    function pin1cb(level)
     du = tmr.now() – pulse1
     print(du)
     pulse1 = tmr.now()
     if level == 1 then gpio.trig(1, "down ") else gpio.trig(1, "up ") end
    end
    gpio.trig(1, "down ",pin1cb)

```

dvv, 2015-02-05:
Neither does `gpio.trig(1, "both", ...)` for me.
2 questions:
i) what are those trailing spaces in strings in your code?
ii) why at all strings are used in this firmware to denote front direction? `gpio.UP` et al. would be more consistent, imho

ystrem, 2015-02-05:
I made little more testing and here it's. The code below works only once then you have to do node.restart() . Node seems operation after the fired trigger. Or this is the right function of gpio.trig?
```
pin = 3
gpio.mode(pin,gpio.INT)
function pin1cb()
     print("HIHIHI")
end
gpio.trig(pin, "both",pin1cb)
```

```
NodeMCU 0.9.5 build 20150127  powered by Lua 5.1.4
> HIHIHI
=node.heap()
22200
> node.restart()
> 7!a!�G8˙A3l�ČVL]˙łá

NodeMCU 0.9.5 build 20150127  powered by Lua 5.1.4
> =node.heap()
22200
> HIHIHI
=node.heap()
22200
> 
```

clixx-io, 2015-02-07:
sometimes you have to combine it with 'gpio.PULLUP' depending on your
hardware

clixx-io, 2015-02-08:
count = 0
delay = 0
gpio.mode(4,gpio.INT,gpio.PULLUP)

function transmit_msg()
   sk=net.createConnection(net.TCP, 0)
   sk:on("receive", function(sck, c) print(c) end )
   sk:on("connection", function(sck) sk:send("ButtonPress\r\n")
tmr.delay(100000) sk:close() end )
   sk:connect(8080,"192.168.0.3")
   print("Sent")
   end

function buttonpress(level)
   x = tmr.now()
   if x > delay then
      delay = tmr.now()+100000
      count = count + 1
      print(".")
      transmit_msg()
      end
   end

gpio.trig(4, "down",buttonpress)

On Sun, Feb 8, 2015 at 10:50 AM, David Lyon <david.lyon.preisshare@gmail.com

ystrem, 2015-02-08:
Thank you ! It's working now. Now it's time for some debouncing :D stuff...

```
pin = 3
gpio.mode(pin,gpio.INT,gpio.PULLUP)

function pin1cb()
     print("HIHIHI")
end
gpio.trig(pin, "down",pin1cb)

closed 2015-02-08 by ystrem