nodemcu/nodemcu-firmware #180: WIFI doesn't auto reconnect

Suxsem, 2015-02-05:
Hi! Maybe I'm missing something but (in station mode) if the wifi router is switched off and then switched on quickly the esp8266 board reconnects to it automatically.
If I leave the wifi router switched off for long time (about 5 minutes) the esp8266 doesn't reconnect anymore.

I found wifi.sta.autoconnect() but i didn't understood if it's enable by default.

MarsTechHAN, 2015-02-06:
You can use 
```lua
wifi.sta.autoconnect(1) 
```
to enable this feature.

nodemcu, 2015-02-08:
autoconnect is enabled by default.
can anyone re-produce this issue?

TheoHarbers, 2015-02-08:
Yes i think so. I saw that when i comment out all wifi lines in my init.lua the esp still connects to the last wifi acces point after reboot.

Suxsem, 2015-02-09:
Sorry guys, was a cheap breadboard that sometimes lose a jumper.
Can be closed :)

closed 2015-02-09 by nodemcu

fixingthingsguy, 2015-11-09:
I am seeing a problem where if I power down my network and wait a few minutes the ESP8266 does not reconnect.
Hardware: Nodemcu dev board ver 0.9
Nodemcu: Private load, from frightanic:
NodeMCU custom build by frightanic.com
	branch: master
	commit: f4fc3b95a86ec98c6de0cfb0c1c424e343297420
	SSL: false
	modules: nohll0
WHAT DOES THE CODE DO?
-It starts up as SOFTAP, then receive STATION ssid/pwd and sets it up. (Thanks to Horacio/Mconnell)
-GPIO0 is set low during above
-GPIO0 is then set high, and device connects to STATION AP.
Then I power down the Station , wait a few minutes and power up, there is no connection established.
If I type in "reconnect" it does reconnect.
The reason for the private load is to include drivers for a display unit.
Any help appreciated.
Code
     INIT.LUA
      pin = 3
      gpio.mode(pin,gpio.INPUT,gpio.PULLDOWN)
      if (gpio.read(pin)==1 ) then print("hell0")
      wifi.sta.connect()
      print(wifi.sta.getip())
      print(wifi.sta.status())
      print(wifi.sta.getconfig())
      else  
      dofile ("modifiedwebpage.lua")
      end
      MODIFIEDWEBPAGE.LUA
--http://www.instructables.com/id/ESP8266-based-web-configurable-wifi-general-purpos-1/
--Code by HoracioBouzas, formatted for use with ESPlorer by jimoconnell
print("WIFI control");
-- put module in AP mode
wifi.setmode(wifi.SOFTAP);
print("ESP8266 mode is: " .. wifi.getmode());
cfg={};
-- Set the SSID of the module in AP mode and access password
cfg.ssid="ESP8266";
cfg.pwd="passwordpassword";
if ssid and password then
print("ESP8266 SSID is: " .. cfg.ssid .. " and PASSWORD is: " .. cfg.pwd);
 end;

 wifi.ap.config(cfg);
ap_mac = wifi.ap.getmac();
-- create a server on port 80 and wait for a connection, when a connection is coming in function c will be executed
sv=net.createServer(net.TCP,30);
sv:listen(80,function(c)
c:on("receive", function(c, pl)
-- print the payload pl received from the connection
print(pl);
print(string.len(pl));  
-- wait until SSID comes back and parse the SSID and the password
print(string.match(pl,"GET"));
ssid_start,ssid_end=string.find(pl,"SSID=");
if ssid_start and ssid_end then
amper1_start, amper1_end =string.find(pl,"&", ssid_end+1); end
if amper1_start and amper1_end then
http_start, http_end =string.find(pl,"HTTP/1.1", ssid_end+1);end
if http_start and http_end then
ssid=string.sub(pl,ssid_end+1, amper1_start-1);
password=string.sub(pl,amper1_end+10, http_start-2);
print("ESP8266 connecting to SSID: " .. ssid .. " with PASSWORD: " .. password);end
if ssid and password then
sv:close();  
     
--close the server and set the module to STATION mode
wifi.setmode(wifi.STATION); 
  
print("ESP8266 mode now is: " .. wifi.getmode());
--configure the module wso it can connect to the network using the received SSID and password
wifi.sta.config(ssid,password);

print("Setting up ESP8266 for station mode--Please wait.");
 

print( wifi.ap.getip());
for i=1,10000 do   
i=i+1
end
print(i)
print( wifi.sta.getip());

end 
-- this is the web page that requests the SSID and password from the user
c:send("<!DOCTYPE html> ")
c:send("<html> ")
c:send("<body> ")
c:send("<h1>ESP8266 Wireless control setup</h1>")
mac_mess1 = "The module MAC address is: " .. ap_mac
mac_mess2 = "You will need this MAC address to find the IP address of the module, please take note of it."
c:send("<h2>" .. mac_mess1 .. "</h2>")
c:send("<h2>" .. mac_mess2 .. "</h2>")
c:send("<h2>Enter SSID and Password for your WIFI router</h2>")
c:send("</form> </html>")
c:send("<form action='' method='get'>")
c:send("SSID:")
c:send("<input type='text' name='SSID' value='' maxlength='100'/>")
c:send("<br/>")
c:send("Password:")
c:send("<input type='text' name='Password' value='' maxlength='100'/>")
c:send("<input type='submit' value='Submit' />")
end);
end);

fixingthingsguy, 2015-11-09:
Sorry, the load info didn't take, here it is
NodeMCU custom build by frightanic.com
	branch: master
	commit:
f4fc3b95a86ec98c6de0cfb0c1c424e343297420
	SSL: false
	modules: nohll0

fixingthingsguy, 2015-11-09:
Sorry, the load info doesn't seem to go in properly?
NodeMCU custom build by frightanic.com
	branch: master
	commit:
commit:   " f4fc3b95a86ec98c6de0cfb0c1c424e343297420"
	SSL: false
	modules: nohll0

TerryE, 2015-11-09:
@fixingthingsguy.  Hello anonymous person.  Please see #719.  You need a dev build and you need to read my unofficial FAQ and research so many other threads where people go down similar potholes.  How the modified webpage examples ever worked is beyond me..

fixingthingsguy, 2015-11-10:
TerryE -Thanks. I did some investigation(esp8266.com forum) and found that I need to insert a disconnect (as in disconnect the link, that was broken by the router being taken off line) and then it reconnects once the router is back on line. For completeness, I have added here the complete code. I have inserted a 
dofile("calls your application that's on a tmr call") and there you have it.
Its a complete set up that
a-allows one to set up a ssid/pwd by connecting to a pc using a low gpio0 setting
b-when powered down and up, the device connects to a router and executes the dofile("calls your application that's on a tmr call") 
c-during (b), if the router is lost and regained, the device reconnects.
d-also reconnects when device is powered down and back up
That's it. 
And I appreciate you folks trying to get nodemcu compatible with all of the sdk features reliably.
Regards
     -- INIT.LUA
      pin = 3
      gpio.mode(pin,gpio.INPUT,gpio.PULLDOWN)
      if (gpio.read(pin)==1 ) then print("hell0")
     wifi.sta.disconnect()
      wifi.sta.connect()
      print(wifi.sta.getip())
      print(wifi.sta.status())
      print(wifi.sta.getconfig())
    dofile(" application that's on a tmr call") 

      else  
      dofile ("modifiedwebpage.lua")
      end
-- APPLICATION THATS ON A TMR CALL.Lua
   function whatever() 
    if wifi.sta.getip== nil then wifi.sta.disconnect() wifi.sta.connect() end
   ---insert your code, such as push button for example
-- if you want to send stuff to a cloud server, this is where it will go, like push button for example
   end
   tmr.alarm( 5,1000,1,function() whatever() end ) 
-- MODIFIEDWEBPAGE.LUA
--http://www.instructables.com/id/ESP8266-based-web-configurable-wifi-general-purpos-1/
--Code by HoracioBouzas, formatted for use with ESPlorer by jimoconnell
print("WIFI control");
-- put module in AP mode
wifi.setmode(wifi.SOFTAP);
print("ESP8266 mode is: " .. wifi.getmode());
cfg={};
-- Set the SSID of the module in AP mode and access password
cfg.ssid="ESP8266";
cfg.pwd="passwordpassword";
if ssid and password then
print("ESP8266 SSID is: " .. cfg.ssid .. " and PASSWORD is: " .. cfg.pwd);
 end;

 wifi.ap.config(cfg);
ap_mac = wifi.ap.getmac();
-- create a server on port 80 and wait for a connection, when a connection is coming in function c will be executed
sv=net.createServer(net.TCP,30);
sv:listen(80,function(c)
c:on("receive", function(c, pl)
-- print the payload pl received from the connection
print(pl);
print(string.len(pl));  
-- wait until SSID comes back and parse the SSID and the password
print(string.match(pl,"GET"));
ssid_start,ssid_end=string.find(pl,"SSID=");
if ssid_start and ssid_end then
amper1_start, amper1_end =string.find(pl,"&", ssid_end+1); end
if amper1_start and amper1_end then
http_start, http_end =string.find(pl,"HTTP/1.1", ssid_end+1);end
if http_start and http_end then
ssid=string.sub(pl,ssid_end+1, amper1_start-1);
password=string.sub(pl,amper1_end+10, http_start-2);
print("ESP8266 connecting to SSID: " .. ssid .. " with PASSWORD: " .. password);end
if ssid and password then
sv:close();  

--close the server and set the module to STATION mode
wifi.setmode(wifi.STATION); 

print("ESP8266 mode now is: " .. wifi.getmode());
--configure the module wso it can connect to the network using the received SSID and password
wifi.sta.config(ssid,password);

print("Setting up ESP8266 for station mode--Please wait.");

print( wifi.ap.getip());
for i=1,10000 do   
i=i+1
end
print(i)
print( wifi.sta.getip());

end 
-- this is the web page that requests the SSID and password from the user
c:send("<!DOCTYPE html> ")
c:send("<html> ")
c:send("<body> ")
c:send("<h1>ESP8266 Wireless control setup</h1>")
mac_mess1 = "The module MAC address is: " .. ap_mac
mac_mess2 = "You will need this MAC address to find the IP address of the module, please take note of it."
c:send("<h2>" .. mac_mess1 .. "</h2>")
c:send("<h2>" .. mac_mess2 .. "</h2>")
c:send("<h2>Enter SSID and Password for your WIFI router</h2>")
c:send("</form> </html>")
c:send("<form action='' method='get'>")
c:send("SSID:")
c:send("<input type='text' name='SSID' value='' maxlength='100'/>")
c:send("<br/>")
c:send("Password:")
c:send("<input type='text' name='Password' value='' maxlength='100'/>")
c:send("<input type='submit' value='Submit' />")
end);
end);

karrots, 2015-11-10:
Not to take this too far into esp8266.com teritorry but you could use the dev branch and convert your code to using the wifi event frame work. When you get to the APNOTFOUND event you could trigger a disconnect. This would replace your timer code.

fixingthingsguy, 2015-11-10:
Jonathan, Thanks for the suggestion. Seems like a better way to manage disconnects. I found several references, please if possible,direct me to the site where I can access. Or can I just request frightanic to build for me. I don't want to bother them if there is a binary lying around.Thanks

karrots, 2015-11-10:
http://frightanic.com/nodemcu-custom-build/

It's anautomated  build service so don't feel like your putting anyone out.

fixingthingsguy, 2015-11-10:
Cool. ThanksSudhirB

fixingthingsguy, 2015-11-21:
All: If you get here, and are interested in end user setup of ssid/pwd, please use a superior solution that is available in dev available through the thread at https://github.com/nodemcu/nodemcu-firmware/pull/644. Implemented by Robertfoss. I have done some preliminary testing with a load by frightanic and it works fine. A really nice implementation.

fixingthingsguy, 2022-02-16:
I'm using the Arduino IDE these days for ESP development. Look in there for the features you want.