nodemcu/nodemcu-firmware #183: DHT22.LUA don't work.

FourQuarks, 2015-02-06:
There is a problem with DHT22.LUA.
I tried the original version without any change.
The line reported (68) contains the lua code "h = b[1] * 256 + b[2]".
The oscilloscope shows that the signal is fine.

What do I wrong?

NodeMCU 0.9.5 build 20150127  powered by Lua 5.1.4
> = node.heap()
22568
> print("DHT22", require("dht22").read(4, true))
dht22.lua:68: attempt to compare number with nil
> =node.heap()
18600
> print("DHT22", require("dht22").read(4, true))
dht22.lua:68: attempt to compare number with nil
> print("DHT22", require("dht22").read(4, true))
dht22.lua:68: attempt to compare number with nil
>

FourQuarks, 2015-02-06:
This line fails: if t > 0x8000 then t = -(t - 0x8000) end

The cause could be this line: t = b[3] * 256 + b[4]

I printed "t" and the output were nil. I don't know why. Maybe a floating point side effect?

   -- lua code --
    if dht22 then
	  print ("2")
    h = b[1] * 256 + b[2]
	  print ("3")
    t = b[3] * 256 + b[4]	  
	  print ("4", t)
    if t > 0x8000 then t = -(t - 0x8000) end
	  print ("5", t)
      ...

-- terminal output --
2
3
4       nil
dht22.lua:79: attempt to compare number with nil
>

myplacedk, 2015-02-06:
That looks more like `yet-another-dht22.lua` than `dht22.lua`.
You can try `nodemcu-firmware/lua_modules/dht22/dht22.lua` and see if that works.

dvv, 2015-02-06:
I see nothing to fail at line 68 of `yet-another-dht22`.
Moreover, from the issue I see one `require`s `dht22`.

We should already conclude on floating point and expose the flag indicating whether firmware is built with or without floating point. I vote pro boolean `node.has_floats` or similar.
Without this it's nearly impossible to write robust modules for this firmware.

FourQuarks, 2015-02-06:
Here is a the file test.lua:
local b = {10, 10, 10, 10, 10}
t = b[2] + b[1] * 256
h = b[3] * 256 + b[4]
print (h, t)

The output on the terminal is:
> dofile "test.lua"
nil     2570
>

The output should be "2570     2570" not "nil     2570". Best is to disable floats!

dvv, 2015-02-06:
Well... Seems like print's clearly broken
What does `print(h==t)` print?

FourQuarks, 2015-02-06:
I think it is more likely an issue with the floating point arithmetic or with the assignment from a table to a variable. 

> dofile "test.lua"
nil     2570
> print (t==h)
false
> = h
nil
> = t
2570
>

FourQuarks, 2015-02-13:
The issue is solved with version 20150212.
Thank you to nodeMCU team.

closed 2015-02-13 by nodemcu