nodemcu/nodemcu-firmware #185: reboot after socket close and connect

vChrisR, 2015-02-06:
When I close a socket and then try to open it again the chip reboots.

I am just testing a connection to nc -l 2000 running on my linux machine.

So this works:
sk=net.createConnection(net.TCP,0) 
sk:connect(2000, "<ip address here>")                                             
sk:send("hello world")

But when I then run
sk:close()
sk:connect(2000, "<ip address here>")                                           

The chip reboots. If I just keep sending data with sk:send() everything works fine.

MarsTechHAN, 2015-02-07:
First, check your version. And update to newest one.
Just reboot without any information?

vChrisR, 2015-02-07:
Version is 0.9.5 build 10250127

Here is the output from ESPlorer:

NodeMCU 0.9.5 build 20150127  powered by Lua 5.1.4
lua: cannot open init.lua
> sk=net.createConnection(net.TCP,0)
> sk:on("receive", function(sck, c) print (c) end )
> sk:connect(2000, "192.168.94.58")
> sk:send("hello world")
> sk:close()
> sk:connect(2000, "192.168.94.58")
c_��RS��FjS��FjS�f��

NodeMCU 0.9.5 build 20150127  powered by Lua 5.1.4
lua: cannot open init.lua

nodemcu, 2015-02-07:
this reboot issue is confirmed. socket can not reused in current version.
in current version, sk:close() will disconnect and destroy the socket connection.
free the memory for tcp conn.
and sk:connect() has a bug, to use a pointer which is freed..
fixed in source code.
for the 20140127 pre_build bin, try:
```lua
sk=net.createConnection(net.TCP,0)
sk:on("receive", function(sck, c) print (c) end )
sk:connect(2000, "192.168.94.58")
sk:send("hello world")
sk:close()
sk = nil
sk=net.createConnection(net.TCP,0)
sk:on("receive", function(sck, c) print (c) end )
sk:connect(2000, "192.168.94.58")
```

closed 2015-02-07 by nodemcu

vChrisR, 2015-02-08:
I build the latest version from source. Works for fine! thanks for fixing this issue!

commit dec47d53 (2015-02-07): fix #185