nodemcu/nodemcu-firmware #189: Specify DNS SERVER ADDRESS

mikewen, 2015-02-07:
dns() function failed quite often.

Is is possible to let dns() use Google DSN servers? ( 8.8.8.8 or 8.8.4.4 )

Also, it seems dns cause memory leaking if it fails.

MarsTechHAN, 2015-02-09:
Thank for your advice. We are discussing about adding some more API in IP level.

Toshik, 2015-02-19:
I think it is better to be able specify own DNS servers to resolve local addresses

jorgempy, 2015-05-16:
hi, i had the same issue, so i took it in my hands,
after adding the get/set dns server, it felt awkward adding it to net, so i added a net submodule "dns"
after that it seemed to me that the present code to resolve dns was not *that* clean, because you are creating a socket that probably don't need, so i coded a second version that creates the socket and "hides it", i think i would be good to deprecate the old version in favor of the new one.

I added pullrequest #423, please take it into consideration, thanks.

the api is:
```lua
net.dns.setdnsserver("8.8.8.8", 1) --sets first dns server
net.dns.setdnsserver("8.8.8.8")  --equal to above
net.dns.setdnsserver("8.8.4.4", 2)  --sets second dns server

net.dns.getdnsserver(1) --gets first dns server
net.dns.getdnsserver() --equal to above
net.dns.getdnsserver(2)  --gets second dns server
```
the amount of dns servers is limited by a config in lwip (modifiable, but 2 seem enough)

the new method of resolving:
```lua
net.dns.resolve("www.google.com", function (ip)
             if ip == nil then print("DNS fail!") else print(ip) end
end )
```

jorgempy, 2015-06-09:
PR #470 should close this

closed 2015-12-14 by nodemcu

tedder, 2017-06-07:
For posterity (ie people who google and find this), `resolve` returns two params, so the new method needs to be changed:

    net.dns.resolve("www.google.com", function (socket, ip)
             if ip == nil then print("DNS fail!") else print(ip) end
    end )