nodemcu/nodemcu-firmware #190: Document somewhare that the MQTT client uses v3.1.1 and not v3.1 of the protocol.

kmpm, 2015-02-07:
I'm fine with using v3.1.1 only but I had to look at the traffic in Wireshark before I understood why I could not connect.
It should have been more obvious to me perhaps but now it wasn't. And I'm probably not alone on using the default mosquitto server in ubuntu 14.04 which doesn't support v3.1.1.

simonvanderveldt, 2015-02-11:
@kmpm I've just added a note to the API documentation, see the [API wiki page](https://github.com/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en#mqttclient). Is that sufficiënt?

(@nodemcu As a non-contributor I can edit the wiki page, not sure if that's supposed to be the case?)

nakatomi845, 2015-02-13:
@kmpm Hi, I've just update my mosquito server to version 1.3.5 on my Debian and I have still problem with connection. Could you write what have you done to run esp with mosquitto?

kmpm, 2015-02-15:
It's quite sufficient.
But I can agree, the API docs should perhaps be in the git repository instead.

closed 2015-02-15 by kmpm

kmpm, 2015-02-15:
Then main thing I noticed was that I couldn't get it to work with a
anonymous connection, I had to login with a, on the server, configured user.
And I'm using 1.3.5 on ubuntu so that should be close enough compared to
your debian.

MQTT_SERVER='192.168.1.101'
MQTT_PORT=1883

m = mqtt.Client(wifi.sta.getmac(), 60, 'user', 'password')
m:connect(MQTT_SERVER, MQTT_PORT, 0)

Bircroad Services | Peter Magnusson | peter@birchroad.net | +46 (0)702 633
276